iNavFlight/inav #4591: No connection diagram is shown on this wiki page

expipiplus1, 2019-04-08:
https://github.com/iNavFlight/inav/wiki/iNav-with-the-FuriousFPV-F35-Lightning-FC

parchedpacha, 2019-04-08:
I looked up the dude who made the first commit to that page, nitr0123, aka Nitro_123 on some other sites, and the person has a real name too, unfortunately this person hasnt made any more commits and has a VERY common name so Idk if youll be able to find this person to ask for that image.

If you manage to figure it out, this would be an excellent way to make your mark on this project.

giacomo892, 2019-04-08:
Page has been removed since it was incomplete and provided no info.
You can link your crossfire receiver to an additional serial port and enable MSP there. Then with bluetooth on the TX you can connect to the PC and configure the FC remotely.
You can also configure your FC using MSP over CRSF for instance and it should work.

closed 2019-04-08 by giacomo892

expipiplus1, 2019-04-09:
Thanks for the info!