iNavFlight/inav #4595: GPS Data Through Telemetry Showing Through Setup Page, but Not GPS Page

Tyson-D, 2019-04-10:
Recently I opened an issue where I was only able to get GPS readings on a wired connection; not through telemetry. I then found out yesterday that on the Ports page, it is necessary to setup  telemetry protocol in addition turning on the MSP function for my 3dr radios (they use Sik 2.0 with MSP). Although I had thought I was in the clear and ready to fly with GPS, an odd issue arose in the configurator. When I connect my ground station to the flight controller via telemetry and go outside, the setup page states that I am able to get a GPS fix, along with my current longitude and latitude (first picture). However, when I head to the GPS page to see where I am on a map, no data is shown (second picture). Does anyone know what could be causing this issue or how to solve it? I have tried using all of the different telemetry protocols in the Ports tab and I got the same result every time, but the setting is currently on MAVLink.

[image]
[image]

teckel12, 2019-04-11:
@Tyson-D do you have an internet connection when outside so it can download the map?

Tyson-D, 2019-04-11:
Yes. I when I run the GPS with the same settings using a wired connection to the flight controller in the same location, the map loads as soon as I get a fix.

closed 2019-06-24 by stale[bot]

OlivierD, 2020-03-24:
Why is this issue closed, what is the actual resolution? I am using INAV 2.4.1, the GPS data shows when I am connected with a USB wire to the board, with a 3D fix, the map shows the paper airplane icon with the correct location. When I switch to a telemetry connection (3DR 915 Mhz), I can see the FC and navigate all the menus and change values but the GPS shows 0 values everywhere thought he health icon up top shows it's blue. In mission control, there is no fix either, just a map. The computer I am using has an active internet connection. It seems the GPS data is not passed over to the telemetry stream. Is this something that needs to be turned on in the CLI?