iNavFlight/inav #4597: Dshot1200 not working lumenier lux f7 + hobbywing xrotor 4 in 1 esc

pediegoivan, 2019-04-12:
Hi all!
Yesterday I was trying out dshot 1200 with a lumenier lux f7 and a hobbywing xrotor micro 4 in 1 45A esc.
When i give throttle to the 4 motors at the same time one of the esc reboots. If I do it with Dshot 600 it just works.
Is this an issue? Or should take this to blheli32 people?

DzikuVx, 2019-04-12:
Hardware issue
https://www.youtube.com/watch?v=TVkBNhii9sw
https://www.youtube.com/watch?v=ELVrn-srnak&feature=youtu.be

closed 2019-04-12 by DzikuVx

pediegoivan, 2019-04-12:
I don't think so,
It is a 4 in 1 esc. The wires length literally don't reach 5 cm.
I can stick my oscilloscope to it if you want.
Why Dshot 1200 is not recomended by the configurator? Just the reasons you give in your video?

teckel12, 2019-04-13:
@pediegoivan You should use Multishot. Or if you're the gambling type, use DSHOT 600.