iNavFlight/inav #4598: inav accelerometer calibration   Matek F405 (new) , Matek GPS

Eski1985, 2019-04-12:
Hi everybody.

Creat this account , becouse i think this is the only place where i can find help.
I know i know that should be moust simple step to do but for some reason i cannot calibrate my accelerometer.

I have got  Matek F405 Wing  ( new ) + GPS matek 
Radio Link Frsky R9M  

[image]

If you take a look on that picture , on the right side is accelerometer is calibrate but on the top is still on red. 

Also when i moving the plane in the Inav my plane (that arrow ) don't move at all. 

Magnetometer is OFF 

any chance someone can help me  ?

Thank You
Eski

DzikuVx, 2019-04-12:
The red mark in the top means that accelerometer was not detected, not that is is not calibrated. Restore defaults, check acc hardware on Configuration page. If still no luck, that might indicate hardware failure

Eski1985, 2019-04-12:
DzikuVX Thank you , it was set up MPU6500 , where should be MPU6000 is working fine ;)

closed 2019-04-12 by DzikuVx