iNavFlight/inav #4599: INAV Gimbal Kakute F7

dominikin9, 2019-04-12:
Hello to all,

I am currently building a quad with a gimbal attached to it for an gimbal.
I am using the Kakute F7 board. 

I have the following problem: 
I want to control the pitch from the gimbal, which needs a pwm signal for that, so I need to forward a channel from my transmitter. I have added a servo via the mixer, but I don't know where I need to solder my connections on.

Does someone know that? 

Greetings Dominic

wx4cb, 2019-04-12:
if you look in the mixer page it will tell you which port to put it on

dominikin9, 2019-04-12:
> if you look in the mixer page it will tell you which port to put it on

I should display it at motor 6 but it doesn't do it
[image]

wx4cb, 2019-04-12:
Your sevo number should be 1 i beleive?

dominikin9, 2019-04-12:
> Your sevo number should be 1 i beleive?
> […](#)

tryed it but doesn't work neither, any other thinks that you think of that might work?
[image]

DzikuVx, 2019-04-12:
I'm affraid Kakute F7 can not drive any servos in Multirotor configuration. In theory it is possible but I do not have the board to check. If you want maybe I can make a test build with 2 servo outputs enabled. But you would have to test if it's working

dominikin9, 2019-04-12:
> I'm affraid Kakute F7 can not drive any servos in Multirotor configuration. In theory it is possible but I do not have the board to check. If you want maybe I can make a test build with 2 servo outputs enabled. But you would have to test if it's working

That would be awesome

DzikuVx, 2019-04-12:
[inav_2.1.0_KAKUTEF7.hex.zip](https://github.com/iNavFlight/inav/files/3074655/inav_2.1.0_KAKUTEF7.hex.zip)

@dominikin9 please check this firmware

dominikin9, 2019-04-12:
> [inav_2.1.0_KAKUTEF7.hex.zip](https://github.com/iNavFlight/inav/files/3074655/inav_2.1.0_KAKUTEF7.hex.zip)
> 
> @dominikin9 please check this firmware

Thank you I will try it

dominikin9, 2019-04-13:
> [inav_2.1.0_KAKUTEF7.hex.zip](https://github.com/iNavFlight/inav/files/3074655/inav_2.1.0_KAKUTEF7.hex.zip)
> 
> @dominikin9 please check this firmware

Thank you it worked now I can control it

DzikuVx, 2019-04-13:
OK, since it workning, i'm closing it. Thanks

closed 2019-04-13 by DzikuVx

arnaudmorin, 2019-08-25:
Hey,
Can you share the patch you did?
I'd like to do the same on current release for the same purpose!

Thanks!

arnaudmorin, 2019-08-26:
nvm, I did no see the related patch yesterday and it seems that is has been merged.
Thanks

geniusboywonder2020, 2020-03-24:
> [inav_2.1.0_KAKUTEF7.hex.zip](https://github.com/iNavFlight/inav/files/3074655/inav_2.1.0_KAKUTEF7.hex.zip)
> 
> @dominikin9 please check this firmware

@DzikuVx 
Hi Pawel,
Can you make this same firmware for 2.4.1 I am trying to use servos on s5 and s6.
I have tried the 2.1.0 hex file and cannot connect into inav after flashing.