iNavFlight/inav #4601: Dshot 1200 does not work correctly in INAV and it does in betaflight

pediegoivan, 2019-04-12:
Yesterday I opened an issue (#4597, now closed)
I confirmed that it is not a hardware issue (the reason of being closed)
I tested DShot 1200 ESC protocol with betaflight and it worked like a charm.
But with INAV, when I raise the throttle using the motors tab, at 1149 all the motors start spining. At 1150 ESC#3 reboots while #1,2,4 keep working perfectly.
Using betaflight it just works, so it must be a code issue.
I am using a Lumenier Lux F7 (matek F722SE target) with a 4 in 1 esc (hobbywing rotor x G2 45A)

teckel12, 2019-04-13:
@pediegoivan Please read the warnings about DSHOT with INAV. I can't express enough that you shouldn't use DSHOT under any conditions. Use Multishot, cause it's reliable. Honestly, I believe DSHOT should be removed from INAV, causes nothing but problems.

pediegoivan, 2019-04-13:
Dshot 1200 is faster than multishot! And has more features. And you know it. I watch all your youtube videos.
The thing is that. If it works in betaflight it should work with inav. Same hardware. I just flashed betaflight in this setup that was not working and in betaflight works.
Just saying, there must be a firmware issue or something.

teckel12, 2019-04-13:
@pediegoivan DSHOT 1200 isn't faster when it sends the same command multiple times to avoid communication problems like BF does. Just use Multishot. INAV is about reliability, not marketing BS.

It works in BF because it send the same command multiple times, negating any speed you believe you have. Again, marketing BS. Digital isn't always better.

pediegoivan, 2019-04-13:
Then .... Delete it from inav right? What is the purpose of having it implemented if there is no point. Like the 32k/32k shit on betaflight. It made no sense so it was removed to avoid this type of pointless issues.

teckel12, 2019-04-13:
@pediegoivan I knew it would be a problem, but it's digital, so it's better!

Also, please note that INAV warns you not to use DSHOT 1200. So not working should be expected. It only works in BF because they make it DSHOT 600+.

But honestly, just use Multishot. Double blind testing I bet you can't tell the difference, until your model falls from the sky, then you know you were flying DSHOT.

pediegoivan, 2019-04-13:
With betaflight you know that is multishot when you have to walk towards your quad to flip it over or when you need your motors to beep because you don't have a buzzer and you can't find your quad.
With smooth long range flying without crashes is pointless and you are right.

teckel12, 2019-04-13:
@pediegoivan If you're crashing about, use Betaflight.  I'd consider INAV and Multishot for those who are trying not to crash.  I haven't broke a prop on an INAV model in 2 seasons.  When I want to crash a lot, I fly Betaflight and DSHOT.

Also, I don't think INAV supports flipping or beepers anyway.  Sounds like if these are must-haves you should fly Betaflight.  Both INAV and Betaflight have their place, for crashing a lot and the latest untested protocols, choose Betaflight (it is beta after all).  For reliability, long-range and trying to avoid crashing, use INAV.

Hope this helps!

pediegoivan, 2019-04-13:
I understood everything. And I dont intend to crash this model. but still. There are advantages on DShot over multishot. And I placed an Issue because there is one. To let know the developers that this is happening and maybe, at some point they can take a look at it. Dshot 1200 is not working, and is not a hardware issue.

teckel12, 2019-04-13:
@pediegoivan And DSHOT 1200 is noted as you shouldn't use it.

digitalentity, 2019-04-13:
I'm on it. Seems we might have a timer issue that only manifests at very high-speed protocols.

digitalentity, 2019-04-13:
Ha-ha! DSHOT1200 is inaccurate on F4 on most timers.

Timer clock for most timers is 168/2 = 84MHz
DSHOT1200 base clock is 24MHz so timer clock divisor is **3.5**!!!
Betaflight rounds the divisor while INAV truncates it which cases the base DSHOT clock to be actually different on Betaflight and INAV: 21MHz on BF vs 28MHz on INAV.

Funny thing is that neither on Betaflight nor on INAV DSHOT1200 is actually 1200kHz clock - it's effectively DSHOT1050 on BF vs DSHOT1400 on INAV.

The problem is that ESCs apparently tweak the code towards BF's implementation which causes incompatibilities.

Fixed by https://github.com/iNavFlight/inav/pull/4607. Note that it doesn't make DSHOT any more reliable in general, but it should at least make DSHOT1200 flyable now.

closed 2019-04-13 by digitalentity

digitalentity, 2019-04-13:
Note that this issue validates DSHOT fragility. Timing error of 0.01 microsecond (1/100000000 of a second) is enough to break DSHOT.

Whereas on Multishot it will only cause 0.2% of throttle error which is beyond being noticeable - during normal stable hover motor throttle jitters by 0.5-1% even on perfectly stable quads.

pediegoivan, 2019-04-13:
Awesome Job! thank you!