iNavFlight/inav #4608: R9 Mini + F411 Wing F Port Flex Firmware

Christianmix, 2019-04-13:
Let's see if anyone can help me .... I have a matek f411 wing with inav_2.1.0_MATEKF411_SFTSRL2 firmware, I'm connecting my F Port pin from the R9 Mini Flex firmware on the pin rx3 which is the same used by sbus, This but I wanted to know how to receive the complete telemetry, the only thing I get is the RSSI in my osd, is the RX3 the correct pin to be used? Well I copied the Inav Lua script for my taranis x9d and I can not run but I do not think any kind of sensor.
I have flex firmware on my taranis x9d as well.

nordicblue, 2019-05-21:
I know this is from a month back, but the issue is that the F. Port (S. Port) needs to be connected to a TX pad, instead of RX pad. I believe that on the F411, you will need to configure it using soft serial, if you are already using the other TX pads for GPS and smart audio, for example.  Check this diagram. The F. Port would be on the same port as the S. Port wire:

[image]

Daniel-1276, 2019-07-17:
Hi, for having FPORT running you should use TX1 pad and you need the following parameters:
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
And besides that, you need to "invert"/ hack your S.PORT/F.PORT output by hardware. There are some instructions on the internet to this. For example see here https://oscarliang.com/uninverted-sbus-smart-port-frsky-receivers/ and https://quadmeup.com/simplest-hardware-inverter-for-flight-controller/

closed 2019-09-29 by stale[bot]