iNavFlight/inav #4609: Representing Crossfire LQ on OSD

MartinHugh, 2019-04-14:
I am passing LQ through to the OSD via the "RSSI Channel" on the Receiver tab.

This works fine except, when the link is down LQ shows 49% on the OSD.

Presumably the channel is defaulting to 1500us.

In Betaflight it is possible to work around this by setting the failsafe value for any channel under the Failsafe tab (Channel Fallback Settings).  So in other words, setting channel 10 to 1000us in Failsafe mode would result in 0% under Betaflight.

How can I achieve this under iNav Configurator, where there does not appear to be a Channel Fallback Settings panel on the  Failsafe Tab.

Thanks.

wx4cb, 2019-04-14:
then you have it setup wrong. you set the LQ in the crossfire module itself not on the transmitter.

MartinHugh, 2019-04-14:
No, I have set it in the crossfire module. It is not set on the transmitter.

The RX Channel Map, has Dst Channel 10 set to LQ.

But will check.  Thanks

closed 2019-06-13 by MartinHugh