iNavFlight/inav #4610: Adding Geo-fence

rcjarvis, 2019-04-14:
## Current Behavior
I'm a drone instructor in a drone pilot training facility. We now use expansive Pixracer because it has geo fence. I would like to push for Matek's f722 Se with gps. The only hurdle is the fact I nav does not support geo-fencing although rth works flawlessly. By accordance to our S.F.O.C. we need to provide max alt and range...

## Desired Behavior
A way to set a maximum distance and altitude  in all flight modes. SO gps and Baro would remain active in background.

## Suggested Solution
Same geo fencing as Eagle tree Vector, Naza  Pixracer.. Sorry, I do not do code so the complexity of this process is not my cup of tea...BUt I know this would be a game changer.

## Who does this impact? Who is this for? Mostly all pilots, with drone regulations restricting our altitudes, here in Canada we are soon to be at 120 meters but right now it is 90m. At the facility we are restricted to 15m and 80m in distance. I had never used this feature before working at the facility but eversince it has proven worthy since I now use it on all my drones that support Geo Fence.

giacomo892, 2019-04-14:
See 
https://github.com/iNavFlight/inav/issues/62
Altitude limitation in progress in: https://github.com/iNavFlight/inav/pull/4442
Closing

closed 2019-04-14 by giacomo892

packetkiller, 2024-05-26:
As the original question was related to not just altitude but Geofencing, I suggest this be reopened and shown as feature added.
The feature that supports geofencing is the "Programming" tab in iNAV. Adding the following commands (from app or cli) for a circular geofence of 1000m and geofence disable at 500m would accomplish all but the altitude issue that is now also supported in iNAV (at least > ver 5 or greater): 
logic 0 1 -1 1 2 17 0 1 0
logic 1 1 -1 2 2 1 0 1000 0
logic 2 1 -1 7 4 0 4 1 0
logic 3 1 2 18 0 0 0 1 0
logic 4 1 -1 3 2 1 0 500 0
logic 5 1 -1 12 4 0 0 0 0
logic 6 1 -1 8 4 4 4 5 0
logic 7 1 6 18 0 0 0 0 0
logic 8 1 -1 1 5 0 0 0 0
logic 9 1 -1 1 5 0 0 1 0
logic 10 1 8 38 0 14 0 1000 0
logic 11 1 9 38 0 14 0 2000 0
In this example CH 14 is an unused CH for the geofence logic.
Reference: [How to get Geofence for any FPV drone? (youtube.com)](https://www.youtube.com/watch?v=qxkhDCLynU4)