iNavFlight/inav #4611: Matek F411 Wing full soft serial 2 port (RX/TX) on S2/S7

SnappyFPV, 2019-04-14:
## Current Behavior
The Matek F411 Wing with SFTSRL2 Firmare has 2x Softserial but just 2x TX no RX/TX

## Desired Behavior
It would be better to have a third "full" serial with RX TX as Serial 2 and leave Serial 1 as TX only. 

## Suggested Solution
The most unused connections on a Matek F411 Wing are S2/S7 solder Pads (PB5/PA15). I don't know if a additional adjustment for the timer needed (S2/S7 are actual on different timer) ? - i just test this with defined S5/S6 (same timer) as Soft Serial 2 RX/TX and it worked great (tested with a GPS connected).

`#if defined(MATEKF411_SFTSRL2)
#define USE_SOFTSERIAL1
#define SOFTSERIAL_1_TX_PIN PA0 // ST1 pad
#define SOFTSERIAL_1_RX_PIN PA0
#define USE_SOFTSERIAL2
#define SOFTSERIAL_2_TX_PIN  PB5 // S2
#define SOFTSERIAL_2_RX_PIN  PA15 // S7
#define SERIAL_PORT_COUNT 5`

## Who does this impact? Who is this for?
Additional CPU based serial port on Matek F411 for example to use ESP32 LORA INavRadar - when UART 1/2 is in use for CRSF/GPS.

SnappyFPV, 2019-04-14:
`const timerHardware_t timerHardware[] = {
    // DEF_TIM(TIM9, CH2, PA3,   TIM_USE_PPM,   0, 0), // PPM IN

    DEF_TIM(TIM3, CH1, PB4,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S1_OUT
    DEF_TIM(TIM3, CH2, PB5,  TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR,  0, 0), // S2_OUT
    DEF_TIM(TIM4, CH1, PB6,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S3_OUT
    DEF_TIM(TIM4, CH2, PB7,  TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO,  0, 0), // S4_OUT

    DEF_TIM(TIM2, CH2, PB3,  TIM_USE_MC_SERVO | TIM_USE_FW_SERVO, 0, 0), // S5
    DEF_TIM(TIM2, CH3, PB10, TIM_USE_MC_SERVO | TIM_USE_FW_SERVO, 0, 0), // S6
    DEF_TIM(TIM2, CH1, PA15, TIM_USE_MC_SERVO | TIM_USE_FW_SERVO, 0, 0), // S7
	
    DEF_TIM(TIM1, CH1, PA8,  TIM_USE_ANY,   0, 0), //softserial_tx2 - 2812LED TIM_USE_LED
    DEF_TIM(TIM5, CH1, PA0,  TIM_USE_PPM,   0, 0), //use rssi pad for PPM/softserial_tx1

    //DEF_TIM(TIM5, CH3, PA2,  TIM_USE_ANY,   0, 0), // TX2
};`

Any changes here needed when PB5 and PA15 are should used as RX/TX?

joeclone, 2019-04-15:
Maybe this #4567

Alejandro-pucci, 2019-04-21:
Hi Snappy,

I am also interested in an extra "full" softserial port. Are you saying you have built a working binary with softserial1 tx on ST1/RSSI pad and softserial2 rx/tx on S5/S6? If so, could you please share it? :) Thanks,

Alejandro

konraddi, 2019-04-23:
Is it technically possible to get another RX/TX interface? I could need it for two F411-Wing builds (FX-61 and Twinstar) that I would like to upgrade with iNav radar but lacking one UART for the ESP32 board (UARTs used by GPS and CRSF).

Many thanks in advance,
Konrad

KorayZ, 2019-04-25:
has this been made into a firmware build or anything? Kind of new to this but want to hook up CSRF for bluetooth connectivitely, only to find out today it needs another uart since i use both gps and csrf. If so, does anyone have the file?

shellixyz, 2019-04-27:
You can try this build with a full-duplex softserial on ST1(TX)/LED(RX): 
[inav_2.1.0_MATEKF411_FD_SFTSRL.hex.zip](https://github.com/iNavFlight/inav/files/3124046/inav_2.1.0_MATEKF411_FD_SFTSRL.hex.zip)

shellixyz, 2019-04-27:
We could use S3/S4 for softserial but then we will need to change stuff in the firmware/configurator so that the displayed output mapping is correct.

SnappyFPV, 2019-04-27:
i tried s2/led but it doesnt work only on s5/s6 (servo 3/4) i think there must be changes in timer also - for s5/s6 timer are the same in code

shellixyz, 2019-04-27:
Using S2 is a bad idea because it uses the same timer as S1. We can't change the timers they are bound to pins and there is only one available per pin for the pins used by this board on this MCU.

Dronek, 2019-05-24:
> You can try this build with a full-duplex softserial on ST1(TX)/LED(RX):
> [inav_2.1.0_MATEKF411_FD_SFTSRL.hex.zip](https://github.com/iNavFlight/inav/files/3124046/inav_2.1.0_MATEKF411_FD_SFTSRL.hex.zip)

@shellixyz could you please contribute a similar FD_SFTSRL version for inavradar? Relates to [this ](https://github.com/mistyk/inavradar-ESP32/issues/12) issue. Would it be possible to include the FD_SFTSRL hex into the standard assets of future inav / inavradar releases?
The SFTSRL2 hex is somehow useless for inavradar when both uarts are occupied.

Dronek, 2019-05-27:
> You can try this build with a full-duplex softserial on ST1(TX)/LED(RX):
> [inav_2.1.0_MATEKF411_FD_SFTSRL.hex.zip](https://github.com/iNavFlight/inav/files/3124046/inav_2.1.0_MATEKF411_FD_SFTSRL.hex.zip)

Please add this as a default asset for future inav releases. Thanks.

konraddi, 2019-05-27:
Is it available for 2.2.0 yet? I would like to use it with Inav radar. Uart
1 = crsf, uart 2= msp on ESP32 and softserial = GPS. This should work?

Thanks,
Konrad

shellixyz, 2019-05-27:
@konraddi @Dronek It is not available yet. I will make a PR for 2.2-RC2

padcom, 2019-06-02:
Should it be possible to use the software serial for MSP over CRSF? I am using it at 57600 bauds (I also tried 19200 but absolutely nothing happens) and the communication stops after receiving the MultiWii API version 2.3.0. Then I get the 10s timeout and that's it.

konraddi, 2019-06-02:
I guess soft serial can do GPS but not crsf. Just use an uart for the crsf
receiver, and plug your GPS on soft serial. That should work in theory, did
that some time ago on a F1 or F3 board...

padcom, 2019-06-02:
Well.. the CRSF receiver is on UART1 and I guess what I was aiming for is to put the CRSF bridge to SoftSerial1 while using GPS on UART2. So I guess the bridged serial over CRSF (not the CRSF connection itself) won't work with SoftSerial1. I'll try what you suggested though.

padcom, 2019-06-02:
So... on that FC GPS on SoftSerial1 works only in NEMA mode :( and with lots of timeouts. I am using BN-880 GPS. Compass works OK on I2C and the GPS starts without any problems on harware UART in UBLOX7 mode. Any hints on how to configure it to make it stable (and possibly in UBLOX7 mode, please)?

Jetrell, 2019-07-09:
Have any of you guys had trouble with the configurator, not reconnecting to the FC after a save/reboot. When using either the, FD SFTSRL or 2 SFTSRL firmware's? 

I don't have this problem when using the standard F411 firmware. 
It almost seems like the extra softserial is messing-up the MSP connection.

guy111a, 2020-10-26:
hello

i am trying to understand how to map s5 or s6 on my matek f411 mini se fc to the soft serial rx .. but i cannot understand who is who on the resource list.

is there a map for this ? somewhere ?

how can i tell what is the soldering pad for A03 ?

thanks.

# resource
IO:
----------------------
A00: FREE 
A01: FREE 
A02: FREE 
A03: SERIAL2 UART RX
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: MOTOR2 OUT
A09: MOTOR3 OUT
A10: FREE 
A11: USB IN
A12: USB OUT
A13: PINIO1 OUT
A14: MPU EXTI
A15: SERIAL1 UART TX
B00: ADC CH1
B01: ADC CH2
B02: BEEPER OUT
B03: SERIAL1 UART RX
B04: MOTOR1 OUT
B05: FREE 
B06: I2C1 SCL
B07: I2C1 SDA
B08: FREE 
B09: SOFTSERIAL1 UART TX
B10: FREE 
B11: FREE 
B12: OSD CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE 
C01: FREE 
C02: FREE 
C03: FREE 
C04: FREE 
C05: FREE 
C06: FREE 
C07: FREE 
C08: FREE 
C09: FREE 
C10: FREE 
C11: FREE 
C12: FREE 
C13: LED1 OUT
C14: LED2 OUT
C15: FREE 
D02: FREE

Jetrell, 2020-10-27:
> 
> i am trying to understand how to map s5 or s6 on my matek f411 mini se fc to the soft serial rx .. but i cannot understand who is who on the resource list.

iNav does not have resource mapping like Betaflight. The softserial pins are in dedicated location.

guy111a, 2020-10-27:
Thank you

So how to know this dedicated location?

Guy A

Jetrell, 2020-10-27:
> 
> Thank you So how to know this dedicated location? Guy A

 From the Matek manual. http://www.mateksys.com/?portfolio=f411-minise#tab-id-3

[image]

guy111a, 2020-10-27:
i saw this .. did not understand.

does it mean that ss1 tx  => st1   rx=> ppm
                             ss2 tx  => tx2   rx=> ? ....

konraddi, 2020-10-27:
What board do you have? F411-Wing?

guy111a, 2020-10-27:
f411- mini se

Jetrell, 2020-10-27:
> 
> i saw this .. did not understand. does it mean that ss1 tx => st1 rx=> ppm ss2 tx => tx2 rx=> ? ....

UART RX2 = ST1 - PPM 
UART TX2 = Softserial (might be okay for VTX power control)

Softserial has to be enabled in the Configurator under - `configuration / other features` and UART 2 turned off under - `Configurator / Ports.`

These SE boards aren't of much use. When you have to trade a UART, for soft serial.

guy111a, 2020-10-27:
Agree :/

Thanks. I managed to put the gps / rx / BT module.. all working perfectly. 

Thanks. 

Guy A

PR #4754 (2019-05-30): Add MATEKF411 target with one full duplex softserial: MATEKF411_FD_SFTSRL
Closes #4611
closed 2021-08-23 by DzikuVx