iNavFlight/inav #4613: Full throttle when barely touching throttle

tott, 2019-04-14:
I'm running Inav 2.1 on a matek 722 mini. Everything works well with the inav configurator connected and when testing motors through the motor sliders but the moment I arm the quad and only go slightly above "motors spinning" the motors spin to full throttle. 
I have air mode disabled and test in acro mode. 

I initially tried this outside but my quad just shot in the air so i had to disarm it. This happened several times before I went back to the drawing board and trying the same while holding the quad in the hand (it's a small 3" so I'm only partially crazy), but I had the same result, the moment motors start spinning a little more than minimum they shoot to full throttle.

My research showed so far that this often happens in Air mode or Angle mode as a result of the quad trying to compensate but not being able to. Given that i am in acro mode without air mode enabled and that all my other quads behave very differently when testing like this I am wondering if there is some compatibility issue with inav or if I just configured something wrong.

Thanks in advance for any ideas and guidance

DzikuVx, 2019-04-16:
possible issues:
- ESC not calibrated
- too much vibrations
- wrongly setup filtering
- bended motor/propeller
- wrong PIDs

CLI `dump` would be useful

tott, 2019-04-16:
Here is the dump from inav cli. Sadly I could not yet figure out how to pull the blackbox data from the quad in linux so this will take a bit longer.
[INAV_cli_ih3_20190416_122612.txt](https://github.com/iNavFlight/inav/files/3084421/INAV_cli_ih3_20190416_122612.txt)

tott, 2019-04-16:
To add to my previous comment.
I calibrated the ESC and it was doing it's song as expected. I tried both DSHOT 150 and 300 and as mentioned the motors spin fine using the motors tab in configurator.
I also made sure my receiver endpoints are properly set.
PIDs/Filters are taken from inav presets for 5" quad.
Rotors look fine. 
As for vibrations - can't really tell - it feels fine until I throttle up.

DzikuVx, 2019-04-16:
I see no obvious problems with the config. Please test in Acro, not Angle mode. Blackbox log will be required to investigate further

tott, 2019-04-16:
Thanks for checking. I definitely tested in acro mode but I'll give it an other shot later and will report back with blackbox logs.

tott, 2019-04-16:
Finally managed to pull the blackbox data. Sadly I don't see anything which makes sense to me. The throttle input as well as the gyro input look good to me in the log file viewer. It would be great if you see something in there.

[blackbox_log_2019-04-16_205828.txt](https://github.com/iNavFlight/inav/files/3086483/blackbox_log_2019-04-16_205828.txt)

shellixyz, 2019-04-17:
You probably didn't wire the motors in the right order or your motors don't turn the right way. Make sure when you test the motors in the motors tab that motor 1 turns when you increase the the first slider, motor 2 with the second one, ... Also make sure the motors turn the right way. Refer to this picture for the motor order and direction of rotation: 

[image]

DzikuVx, 2019-04-18:
Definetly check motor ordering and rotation direction like on a picture above

tott, 2019-04-19:
Thanks for the feedback - I verified the motor directions and order in the configurator and everything looks good :( Do you have any other ideas?

shellixyz, 2019-04-19:
Are you sure your board orientation is set correctly ? Check in the setup tab that the 3D model moves in the same way as you quad

tott, 2019-04-19:
Yes, I did this as well. In the configurator everything works as expected. What really confuses me is that this even happens in acro mode where I would expect the accellerometer to not effect the behavior of the motors, right?

shellixyz, 2019-04-19:
Yes in acro the acc is not used, the gyro is though. So if the alignment is not correct it will impact acro too.

shellixyz, 2019-04-19:
The next step is to check that your gyro is working correctly from the sensors tab

tott, 2019-04-19:
Ok, will do, thanks a lot for all the input.

shellixyz, 2019-04-19:
If the gyro works ok then it is probably just too high PIDs (defaults ?) for your setup

tott, 2019-04-19:
Yes, it's a modified iflight ih2 quad which i upgraded the electronics off and i just loaded the presets for a 5" quad, so the PIDs could be off. Sadly I'm still pretty new to this so I kept my fingers off the PIDs completely :)

Troynica, 2019-04-27:
Are you using Futaba? If that is so, you may want to verify if throttle is not inverted.

closed 2019-07-10 by stale[bot]

quetiano, 2022-06-16:
Hello and good day Mr Tott
Have you tried disabling/configuring Airmode ?