iNavFlight/inav #4614: Short control freeze and RTH did not turn on

Andrey-Prikupets, 2019-04-15:
I observed strange Failsafe behavior on plane and copter. Copter is mine and I have blackbox log of the event. Here is the description of it: after line of sight was suddenly blocked by trees the copter stopped respond to control. I was flying in ACRO and saw that it started to dive and cannot do anything. It lasted for couple of seconds, and then control restored and I saved it from the fall. RTH did not turn on.
From the log, at 1:07.769 all control graphs became straight lines - it is the moment when control froze. RSSI (red line) was still high, and started to drop in 1:08:748 and till to zero.
At 1:09:015 iNAV triggered rxSignalReceived flag (yellow line). failsafe_delay should start counting from this moment, as I see from the source, and it was big enough (20), so when signal was restored at 1:09:786, RTH did not had a chance to turn on.

RX: FrSky D4R-II, D16, No Pulses, CPPM connection.

[image]

Log (available for 7 days): https://wetransfer.com/downloads/c6e4f50c6f268ecc7b704f6f3b673bf620190415202438/ef270818dfaa2af1a606bbc1b191931320190415202438/772641
Dump: [dump.txt](https://github.com/iNavFlight/inav/files/3082072/dump.txt)

serg-ku, 2019-04-15:
Also observed same behavior on two different aircrafts(plane and quadcopter), both were using different tbs crossfire rx and tx, both rx were configured to failsafe "cut" mode. FS didn't fire, and input from tx was updating each 2-3s, between updates it feels like it was holding last stick and switch values. Total time in this state without FS was about 15-20 seconds.

digitalentity, 2019-04-18:
PPM receivers don't report failsafe, so unless they are configured to "NO PULSES" the `rxSignalReceived` will never be set to `false`. As far as PPM driver is concerned receiver signal IS valid.

The yellow line on your graph is `rxFlightChannelsValid` which indicates validity of stick channels. 

Your config sets `set failsafe_delay = 20` which means your receiver should report signal loss for 2200ms = 2.2 seconds before INAV considers it a failsafe. In your log the period of signal loss was ~0.7 seconds so failsafe was not triggered, so this is working as intended.

Closing this as a configuration/setup issue.

closed 2019-04-18 by digitalentity