iNavFlight/inav #4625: Small oscillations on pitch axis in poshold cruise mode

DiscoMan18, 2019-04-19:
Inav 2.1.0, Matek F405-CTR, 450-sized quad. 

Default navigation PIDs. 

When flying straight forward, small oscillations are seen on a pitch axis. This happens only in Poshold with user control mode set to CRUISE. If you set user control mode to ATTI, the quadcopter flies silky smooth, so the problem is obviously in navigation pids. What nav pid should I change to avoid these oscillations? 

Also issues mentioned here [https://github.com/iNavFlight/inav/issues/3706](url) are still not resolved. Sometimes when doing yaw turns (from 90 to 180 deg) the quadcopter starts to freak out of itself, just as described in the issue I posted earlier (check out the link). I doubt it's a tuning issue, as I've already tried all possible combinations of nav pids trying to "ease out" the reactions of the quad. 

After thorough testing I have a strong sensation that velocity pid controller used in GPS Cruise mode is totally buggy and not reliable. When, for example you start to just fly straight the quadcopter first banks slightly left then right, and this effect is much more pronounced at maneuvering when you gain some speed. At 20kmh it's all basically ok, but close to 40-50 kmh I may get a lot of surprises. 

Does anyone have a video example of a quadcopter flying smoothly in GPS Cruise mode? I looked on youtube, but there is almost nothing on this.

shellixyz, 2019-04-27:
You can try to play with `nav_mc_{pos,vel}_xy_{p,i,d}`. These are the settings that control the position in nav modes

DiscoMan18, 2019-04-28:
Already did that thousands of times, no luck. Don't know what to try next. Do you have a video example of a quad smoothly flying in gps cruise mode? I continue to suspect, that this mode still needs to be improved, as its current implementation is still far from perfect... Saying this I strongly believe that's definitely not a tuning issue, cause as I've already said in previous report of a quad behaving strangely in gps cruise mode, I spent the whole summer of 2018 fiddling in all possible ways with gps pids and other settings of inav, but achieved absolutely no result...

shellixyz, 2019-04-28:
> Do you have a video example of a quad smoothly flying in gps cruise mode?

I don't have one myself. Maybe @DzikuVx do. Personally I only used iNav on fixed wing aircrafts until now. I just finished buidling a quad I'm going to be using to test iNav so I will test PH cruise and report soon.

DiscoMan18, 2019-05-16:
Also in order to illustrate the issues mentioned here and in [https://github.com/iNavFlight/inav/issues/3706](url) here is the video, showing bad flight performance in PH Cruise:

[https://www.youtube.com/watch?v=c_ge14j56AQ](url)

eifel2006, 2019-05-23:
Have the same problem. Looks like nodding your head. Small gimbal does not fix it. Already have the same problem as #3706. Matek405ctr 2212/980, dji9443, 380mm frame

DiscoMan18, 2019-07-22:
Does anyone experience similar problem when flying a quadcopter in gps cruise mode?

closed 2019-10-04 by stale[bot]