iNavFlight/inav #4627: roll rate in angle mode

MartinHugh, 2019-04-21:
I have recently set up a 1m fixed wing with an F411-Wing and iNav 2.1

I have trimmed it out and completed servo AutoTrim and Autotune procedures.  However when I bank hard (I am intentionally limited to 60 degrees) the aircraft is very sluggish to start rolling.

Can you please advise where I start in order to rectify this.  Can I simply override a default rate setting, or do I need to tune PIDs ?

shellixyz, 2019-04-22:
Bank rate is limited by the set `roll_rate` and is calculated like this: roll_rate = angle_error * `fw_p_level`. So if your roll rate is set correctly and you still have too slow bank rate increase `fw_p_level`. The default should be enough though, it is already too high for fixed wing aircrafts IMO.

MartinHugh, 2019-04-22:
Thank you @shellixyz, I will look into that.

MartinHugh, 2019-04-22:
I see that fw_p_level is the default of 20 on my setup.

Is there any way I can change this via the OSD menu rather than have to drop into a CLI at the flying field ?

shellixyz, 2019-04-22:
Yes you can change it from the OSD menu.

roll_rate is for all the modes but manual. 28 = 280°/s should be fine
manual_roll_rate = roll max servo travel % in manual mode

MartinHugh, 2019-04-22:
I cannot see anything in the OSD menu which correlates with fw_p_level (20 (200 degrees/s))
Which setting is it ?
Thanks.

Jetrell, 2019-04-23:
> Bank rate is limited by the set `roll_rate` and is calculated like this: roll_rate = angle_error * `fw_p_level`. So if your roll rate is set correctly and you still have too slow bank rate increase [...]

@shellixyz  Can you give Martin and myself, your impression of how reactive to stick input, ANGLE mode should be, with e.g `roll rate ` 35deg and `pitch rate `set to 60deg and the` fw_p_level` set to 25 ?

The reason I ask...  Is because I too have had issues with the F411-wing in conventional air frames. i.e. Standard or V-tails. When it comes to getting the ANGLE mode to react to stick movement in a smooth or even some what linear manor.

I have gone through the settings you have stated, from the Wiki, many times. And can not get ANGLE mode to feel even half confidence inspiring, to fly with. It requires almost full stick to get either pitch or roll to start to react; and even then its very sluggish.
So I just gave up on it with the F411-Wing in those type of air frames and have stuck with HORIZON mode. Which has stacks of authority. 

As a side note. 3CRS also lacks the same authority as ANGLE mode does. Yet the RTH auto navigation control has heaps of authority(from observation). Can I assume it uses the same setting as ANGLE and 3CRS?

Here are my relevant F411-Wing setting on a Nano Talon

```
set fw_p_level = 24
set fw_i_level = 4
set fw_d_level = 62
set max_angle_inclination_rll = 480
set max_angle_inclination_pit = 380
set fw_iterm_limit_stick_position =  0.300
set yaw_p_limit = 275
set nav_fw_pos_z_p = 41
set nav_fw_pos_z_i = 6
set nav_fw_pos_xy_p = 77
set nav_fw_pos_xy_i = 4
set nav_fw_pos_xy_d = 7
set tpa_rate = 40
set tpa_breakpoint = 1600
set fw_tpa_time_constant = 1000
set roll_rate = 34
set pitch_rate = 16
set yaw_rate = 8
set manual_rc_expo = 36
set manual_rc_yaw_expo = 0
set manual_roll_rate = 95
set manual_pitch_rate = 90
set manual_yaw_rate = 80
```

MartinHugh, 2019-04-23:
@Jetrell thanks for your comments on this.

@shellixyz, sorry if my question about the OSD is confusing.  the roll_rate setting is not the "roll_rate", because that is a live calculation based or error and fw_p_level.  I think the roll_rate setting is in fact the MAXIMUM roll_rate permitted.   I do not wish to change that at the field particularly, I need to change the underlying fw_p_level. Is that possible from the menu/OSD.  Thanks.

shellixyz, 2019-04-23:
@MartinHugh  @Jetrell 

> I cannot see anything in the OSD menu which correlates with fw_p_level (20 (200 degrees/s))
> Which setting is it ?
> Thanks.

It should be in the `PID TUNING` submenu. In `OTHER PP`, the `ANGLE STR` setting (angle strength)

> Can you give Martin and myself, your impression of how reactive to stick input, ANGLE mode should be, with e.g roll rate 35deg and pitch rate set to 60deg and the fw_p_level set to 25 ?

It should be very reactive. Even too reactive with `fw_p_level` set at 25, meaning that it would most likely induce oscillations. At least if your pitch/roll PIDs are set correctly.

> As a side note. 3CRS also lacks the same authority as ANGLE mode does

That's very strange at least for roll because for roll 3CRS IS angle mode as soon as you move the roll stick.

@Jetrell 
> set fw_i_level = 4

You lowered that from the default 5. It is not what you want if you want to increase reactivity of angle. `fw_i_level` is angle mode's low-pass filter cut-off. The lower this is the slower it will be to react.

Are your planes flying like you want in acro/rate mode ?

MartinHugh, 2019-04-23:
Thanks for your response @shellixyz 

I currently only have angle mode (and manual) setup, so cannot confirm anything about other flight modes.  There is definitely something not right the roll rate in angle mode, but I will try increasing the fw_p_level next time I am at the field.

MartinHugh, 2019-04-23:
@shellixyz, sorry if I am being thick and missing something obvious here, but in the PID TUNING menu, there is no "OTHER PP" submenu. The closest is a "FILTER PP" submenu.

What am I missing ?

Jetrell, 2019-04-23:
@shellixyz  These are the PI and FF gains on that same Nano Talon.  It is the one that has the least stick authority in ANGLE mode.
```
set fw_p_pitch = 42
set fw_i_pitch = 32
set fw_ff_pitch = 12
set fw_p_roll = 23
set fw_i_roll = 26
set fw_ff_roll = 3
set fw_p_yaw = 0
set fw_i_yaw = 0
set fw_ff_yaw = 0
```
These Feed Forward settings may seem strange to you. But increasing the P gain and lower the FF was the only thing that would stop control oscillations on the PITCH and ROLL axis's, in the wind or at any speed. 
This little plane was a nightmare to set up and get right. It now fly's very well with these setting, with NO YAW stabilization at all. Just YAW manual control.
I literally spent days tuning this air craft. And nearly crashed it heaps of time throughout the tuning process. It now handles speeds of up to 135kph and 20+Kph winds with no handling issues. (flat wing mod).
Now its only issue, is its lack of stick authority to ANGLE bases flight modes. Yet as I said earlier... It reacts very fast (from DVR observation) when under `RTH` of `Pos_hold `auto navigation control. 

@shellixyz _You lowered that from the default 5. It is not what you want if you want to increase reactivity of angle. fw_i_level is angle mode's low-pass filter cut-off. The lower this is the slower it will be to react.

Are your planes flying like you want in acro/rate mode ?

I lowered the `fw_i_level` because it seemed to be fighting a little in `POS_HOLD` mode.
Yes. In acro/air modes it performs very well. As they do in Horizon mode. 

Here is a full settings dump of the Nano Talon -
[Nano Talon F411-Wing (2.1 Full version)Diff all.zip](https://github.com/iNavFlight/inav/files/3109842/Nano.Talon.F411-Wing.2.1.Full.version.Diff.all.zip)

Maybe something stands out to you Michel.

I get the impression that the issue Martin and I are having. Is something to do with F411_Wing. But I can see how! 
 I have not had these problems with other FC's. But I also don't have another FC in a Nano Talon either. I would like to know the percentage of people that are using the Matek F411_Wing in flying wings, compared to conventional tail or V-tail aircraft.

@MartinHugh What air frame are you using?

shellixyz, 2019-04-23:
@MartinHugh Yeah sorry I only checked the code. The submenu is there but has never been enabled for a reason unknown to me. Something we could enable in the next release. You have to use the RC adjustments or the configurator for now to change `fw_p_level`.

shellixyz, 2019-04-24:
@Jetrell Indeed these settings for FF seem very low. Usually FF should be higher or close to P. The idea is that FF is like the "manual term". In acro you input a command that's mapped to a rate between 0 and the max rate you set and that's directly translated to an elevon/aileron/elevator deflection according to the FF setting. If FF is set correctly and your plane is trimmed it should fly well in calm conditions with P and I set to 0. P and I are mainly there to stabilize the plane against external influence (wind) and to correct for small errors in the other tuning parameters (FF, TPA, ...).

> I lowered the fw_i_level because it seemed to be fighting a little in POS_HOLD mode.

fw_i_level is not the right parameter to tune for PH.

You should only change settings directly related to the mode you're tuning.

The flight modes are built on top of each other so you should tune the foundations first. The pyramid is build like this: manual < acro < angle < alt hold < nav modes (controlling position).
So first you need to make sure your plane is set up like you want and is flying as expected in manual mode (mainly setting the throw and trimming the control surfaces). Then you can start tuning acro with fw_{p,i,d}_{pitch,roll,yaw} and {roll,pitch,yaw}_rate. Once it feels good in acro you can start tuning angle with fw_{p,i}_level. But never go back and tune the acro settings to solve issues in angle. Same thing for the other modes. Horizon mode is a mix between angle and acro so if you tuned both acro and angle like you want it should fly good in horizon. Once you tuned angle you can start tuning alt hold with nav_fw_pos_z_{p,i,d}. But never change angle/acro settings for tuning alt hold. Then you can tune the position controller used for PH, RTH, CRUISE with nav_fw_pos_xy_{p,i,d} but I never found that to be needed.

Did you follow the [INAV PIFF controller tuning guide](https://github.com/iNavFlight/inav/wiki/Tune-INAV-PIFF-controller-for-fixedwing) on the wiki ?

Personally I'm using between 14 and 16 for fw_p_level and all my planes are very reactive in angle mode.

I don't see anything special in you diff besides the mentioned low FF factors for pitch and roll. Also the asymmetrical settings for the servo throws but that's not the cause of your issues.

> I get the impression that the issue Martin and I are having. Is something to do with F411_Wing

The F411-Wing is most likely not the issue unless you have a defective gyro/acc but that would be surprising. Most likely a setup issue.

I'm using a F411-Wing FC in my Mini AR wing. Nothing to complain about it is working like any other FC for me.

Here is the profile diff or my Mini AR so that you have something to compare to. Notice the much lower P compared to FF and also the reduced fw_p_level.

```
set fw_p_pitch = 26
set fw_i_pitch = 24
set fw_ff_pitch = 92
set fw_p_roll = 28
set fw_i_roll = 20
set fw_ff_roll = 46
set fw_p_level = 14
set max_angle_inclination_rll = 600
set max_angle_inclination_pit = 600
set tpa_rate = 33
set fw_tpa_time_constant = 600
set rc_expo = 30
set roll_rate = 30
set pitch_rate = 15
set manual_rc_expo = 30
```

Jetrell, 2019-04-24:
Thank you for the in depth reply Michel. Its always good to get extra info.

The answer to your above question -

I have followed the Wiki tuning guide and have tuned many planes before as well. 
But the Nano Talon/ F411-Wing combo, is a BIG exception.

Like you stated on the servo setup throws. Especially on the V-tail.
The Nano Talon causes some strange effects with its mechanical couplings and the V-tail mix itself.
These odd looking figures, are what it took to get the Rudder and Elevator throws perfectly centered and giving equal up and down travel. (mechanical linkages are centered)
It was also why I had to get rid of YAW stabilization.  This plane would never fly straight. Even in 3D Cruise.
 It was a strange YAW / ROLL coupling effect that was taking place. After all.. V-tail Yaw does cause some Roll as well.
The gyro would sense that the ROLL axis needed correction after a YAW axis correction. This would cause slow IMU drift, and make the plane fly around in large circles and never fly straight, in any stabilization mode.
I was at the end of my tolerance with this plane, before I turned off YAW stabilization.  And that made all the difference.

It would take too long to explain the way this plane reacted to normal P and FF on the Roll and Pitch axis.
I have never encountered anything like it; even on my XUAV mini Talon. 
But I'm not sure if the F411_Wing is also causing some issues...Or any offset of the IMU chip, to the plane central axis points, may have some effect.. 

I will load up some of your diff settings and see if it makes any difference.  Although. I haven't had any problem with Flying Wings. ANGLE mode seems responsive with them.
Thanks again for you time @shellixyz 

@MartinHugh Martin. I didn't mean to hi-jack your post.  I hope you have been able to pick up some things from this, that will help you.

MartinHugh, 2019-04-24:
@Jetrell the airframe I am using is a Flying Wings Falcon Evo.   It is about 1m span.

Not sure where to go next apart from increasing fw_p_level.  I will set it from 20 to 23 and try that, although  @shellixyz is suggestion this is likely to be approaching oscillation - so will try it and see.   If anyone can spot anything, I will paste in my diff.  Would appreciate it if there are any "sore thumbs" sticking out in this

DzikuVx, 2019-04-24:
Since I do not see anything that we should change in the firmware over here, I'm closing this one and you can continue discussion in a closed ticket if you really want. Thanks

closed 2019-04-24 by DzikuVx

MartinHugh, 2019-04-24:
@DzikuVx  I was not asking for a firmware change.  I was not reporting a bug.   I am asking for help setting up iNav.  I have followed all the documented procedures.

The advice I have so far from this thread is to change fw_p_level, but with the rider that that is probably not going to help since it is way to high for fixed wings at its default anyway.

I dont see how closing the ticket resolves anything apart from hiding things you dont want to read.

Jetrell, 2019-04-24:
@MartinHugh  I'm sorry about that. It was not my intent to get in your way. I thought that the issue we are both having may have been related and adding more information may have gotten to the bottom of the problem.
May be @DzikuVx read your question wrong and closed it for that reason.
I had a look through your diff settings. And it looks fine to me. Just like mine seems fine as well.

The advice @shellixyz gave us, works for most cases. While in our cases. There appears to be something else affecting it as well.

MartinHugh, 2019-04-24:
@Jetrell no problem, there was a point after your first post when I thought it was helping to focus the problem, but I think after the second, long post it had actually diluted my problem :).  Apart from yourself, I was unable to get an opinion on my diff before the thread was shut down.  Perhaps I should have posted the diff first.

Anyway no problem from my end - except that is for over zealous admins closing things off before they have concluded.

I have changed my fw_p_level from 20 to 23 and will see how that changes things.

I hope you get yours sorted too, and would be interested to hear how you get on.  Good luck with it and thanks for your input anyway.

Jetrell, 2019-04-24:
@MartinHugh  Write back here after you test `fw_p_level ` at 23. I'll be interested to hear your results.

shellixyz, 2019-04-27:
The thread isn't really shut down. GH issues are meant to track bugs and feature requests. If we end up finding a bug or decide something needs to be improved we can reopen it. The thing is that angle mode is very very simple and up to now all the issues that have been brought up on the subject have all been setup issues. Again angle is very simple it just looks for the error between the target angle given by the roll/pitch channels of the receiver and the actual attitude of the wing. Then the error is multiplied by fw_p_level to get the requested angular rate to correct. So unless there is a bug which is unlikely the issue can only be two things: bad attitude estimation or badly setup rate controllers.

@MartinHugh I did take a look at your diff but there isn't much to gain from me looking at it. PIFF settings will be widely different between aircrafts. Only you are able to know if the settings are correct or not. Angle mode only relies on the rate controller setup and fw_p_level. To me if your rate controllers are setup correctly the aircraft should be very reactive in angle with the default fw_p_level of 20. As I said for me and the very varied types of wing/planes I tested iNav on it has always even been too much. Causing very sharp movements and some bounce back if not moving the sticks smoothly.

MartinHugh, 2019-04-27:
@shellixyz thank you for replying.

I hope to fly tomorrow and have set fw_p_level to 23 to see how it changes things.  I will report back any relevant info.

MartinHugh, 2019-04-28:
I can report that flying with  fw_p_level to 23 rather than 20, did give some improvement, but it is still not where I would like this.  It is much better if I feed in full deflection on the stick (i.e. it flies a bit like there is a huge chuck of expo applied to the channel)

Note that there is no sign of any perceptible "proportional" based oscillation, so I think I am safe to crank it up another 3 or 4 points.

shellixyz, 2019-04-28:
Crank it as high as you need to. You can switch back to acro or manual if you start to get oscillations anyway. A good way to test/adjust fw_p_level is through RC adjustments

MartinHugh, 2019-04-28:
@shellixyz, thank you, and noted, will carry on tweaking this setting and see where I get to.