iNavFlight/inav #4631: Nav althold issue

Magicman99, 2019-04-22:
HI New to INAV, attempting to set up Nav ALTHOLD, I can arm and fly without issue when I put the quad into Althold mode it immediately goes to full throttle and stays until I change out of atlhold with no throttle input from myself.  I have changed the hover throttle to as low as 1300 and still the same issue.  I am using a lux F7 with a gps and magnetometer all working and showing appropriate change in the sensor tab. thanks for any help

Magicman99, 2019-04-23:
CLI Dump

# version
# INAV/MATEKF722SE 2.1.0 Feb 25 2019 / 17:01:07 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix
smix reset

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
[... 930 lines cut ...]
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 50
set pitch_rate = 50
set yaw_rate = 45
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100
set fpv_mix_degrees = 0

# battery_profile
battery_profile 1

set bat_cells = 4
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 850
set battery_capacity_warning = 170
set battery_capacity_critical = 85
set battery_capacity_unit = MAH

teckel12, 2019-04-23:
@Magicman99 Have you actually measured your throttle value to set? It could be that nav_mc_hover_thr needs to be 600 for all we know.

Magicman99, 2019-04-23:
yea I did it was around 1500 its a heavy little bugger but by manual throttle control it was about 1500.  I just did another test flight to see if I could get anything out of the black box and the first one again flew to the moon but the next seam to go to where I had set the throttle point.  it drifted down but I took it too far down just to make sure I was not doing thing wrong. I will try it again tomorrow and hopefully it fixed itself.  thanks for the reply

Magicman99, 2019-04-23:
so this may be a really stupid question but is the throttle scale on this 1000-2000 like Betaflight?

teckel12, 2019-04-23:
@Magicman99 Yes, I was being over-dramatic.

Magicman99, 2019-04-24:
did a second test flight today seams to have self corrected not sure what I did but its working now thanks for the help

closed 2019-04-24 by DzikuVx