iNavFlight/inav #4632: Adjustable decimal places for current on OSD

swissfreek, 2019-04-23:
## Current Behavior
Currently, the amp draw readout in the OSD is done to 2 decimal places (x.xx). On my build, the hundredths place (0.0x) fluctuates rapidly, and never settles, which is distracting in the OSD and not useful information. The other digits are steady.

## Desired Behavior
It would be nice if the number of decimal places were selectable in the configurator like they are for the voltage readout. Knowing my current draw to .1 amps is useful, knowing it to .01 amps is not really.

## Suggested Solution
Implement a selection in the OSD tab of the configurator similar to Voltage or Lat/Lon that allows the user to select how many decimal places to use on the current readout.

## Who does this impact? Who is this for?
I suspect that knowing current draw to 0.01 amps is not particularly useful to most users, and adds unnecessary information, clutter, and distraction to the OSD.

## Additional context
None.

teckel12, 2019-04-23:
@swissfreek I can't imagine that current down to the hundredth of an amp is ever useful. In other words, seems it could just be changed to the tenth of an amp in all cases.

swissfreek, 2019-04-23:
I would tend to agree, but who knows, maybe someone has a weird use case?

teckel12, 2019-04-23:
I have 1s tiny whoop quads and a tenth of an amp is plenty low.

shellixyz, 2019-04-23:
Let's make it configurable. Personally I prefer to have 2 digits. The second digits shouldn't be all over the place at least since ADC sample averaging have been introduced if you don't have a noisy measurement.

closed 2023-10-27 by DzikuVx