iNavFlight/inav #4637: IST8310 not detected by iNAV 2.1

PanayiotaCy, 2019-04-24:
For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)

PanayiotaCyI, 2019-04-24:
Hello everyone,

I am trying to interface GPS & Compass module IST8310 (LeadingStar Ublox M8N GPS with IST8310 Compass Module for F35 Inav F4 Flight Controller) with iNAV version 2.1 (configurator 2.1.4), but it is not detected. My flight controller board is MATEK F405-CTR.  According to specs, they should be compatible.  Any help would be greatly appreciated.

shellixyz, 2019-04-27:
This is a known issue #4498. It will be fixed in 2.2. In the meantime use this build: [inav_2.1.0_MATEKF405.hex.zip](https://github.com/iNavFlight/inav/files/3124120/inav_2.1.0_MATEKF405.hex.zip)

closed 2019-04-27 by shellixyz

PanayiotaCy, 2019-05-06:
Dear shellixyz,

Thank you so much for your reply! I will use the suggested go-around for now.