iNavFlight/inav #4638: How do I make it so that throttle is passed through in manual mode when not armed

mbainrot, 2019-04-24:
I had a bit of a look around (though I have no idea what it would actually be called) there doesn't seem to be a way of enabling throttle passthrough in manual mode with AP disarmed. 

Is there a setting I can change which would override this behaviour? (I use throttle cut on the TX as my main "safety" switch)

What I am trying to achieve is a situation where the FC just ends up being a fancy receiver as a last resort if I run into issues.

Model type: Airplane w/ differential thrust, y-splitter ailerons, std elevator/rudder
FC: Open Pilot Revolution
GPS: u-blox SAM-M8Q (Sparkfun 15106)
RX: Spektrum Satellite

giacomo892, 2019-04-24:
You cannot.

>What I am trying to achieve is a situation where the FC just ends up being a fancy receiver as a last resort if I run into issues.

Which kind of issues are you thinking about?

shellixyz, 2019-04-27:
Why not just arm the FC like everybody ?

DzikuVx, 2019-05-09:
You can pass it using servo mixer by assigning throttle input into and servo of your choosing

closed 2019-05-09 by DzikuVx