iNavFlight/inav #4641: Compass movement is strange

Tillidan, 2019-04-24:
Dear all,

I have a F722 flight controller wired with a M8N GPS + Compass module (buyed in Banggood). I have installed last stable INAV firmware and configured basic parameters. In addition I have wired a Bluetooth module, a led strip and a PPM rx module.

Everything have connection and blue icon in INAV. GPS signal is ok also.
I have calibrated several times acelerometer and Compass with battery connected (compass is external in GPS module, it is the HMC5883).

Movement in relation with acelerometer is ok. The strange thing is if I move the drone in Roll or Pitch angle, "direction of drone" in INAV changes, it is like some bad data of Compass. By the way if I only rorate in YAW angle, Compass signal /movement is ok.

I configured the magnetometer in CW 0ºflip. Also I wired 2 pull-up resistences in SDA-SCL from 3.3V (lik you recomended in this forum) but nothing changed.

Could be a faulty Compass?

Thanks in advance!!
Regards.
Javier.

giacomo892, 2019-04-24:
>I configured the magnetometer in CW 0ºflip. 
I'd check if this is correct

Tillidan, 2019-04-24:
I checked it. I have tried all this setup options. It seems it is another thing. Maybe buy another GPS compass module...

giacomo892, 2019-04-24:
It must be something like that or calibration issue

Troynica, 2019-04-27:
Similar to what I once experienced. I don' t know what GPS module you have, but mine was one of those black little domes. It hat an AKM8975 magnetometer chip but an emulator chip presented it as an HMC5883 to the FC. This never worked well because the compass stopped working after some minutes in the air and made the quad fly away suddenly in an unpredictable direction.
You can remove and bypass the emulator chip (if, ofcourse, this applies to your GPS module) so your FC sees an AKM8975, what is actually is. I had to include AKM8975 support for my board to iNav and recompile it because it wasn't included by default. I don't remember what the exact orientation was that I entered. I do remember that it took some time to figure out because the compass really needed calibration before it worked (more or less) correctly.

closed 2019-07-10 by stale[bot]