iNavFlight/inav #4645: tricopter cc3d revo

ricardovalmeida, 2019-04-25:
hi dear friends
I mounted a tricopter using the cc3d revo board.
I wish I could use the s4 output to trigger a location buzzer. Output s4 does not accept servo mix, motor mix only. this can be changed to use the s4 output with servo mix.
thank you.
Ricardo Almeida.

DzikuVx, 2019-04-25:
No, this is not possible, pins can not be assigned at will. 
Full explanation over here https://www.youtube.com/watch?v=v4R-pnO4srU

closed 2019-04-25 by DzikuVx

ricardovalmeida, 2019-04-28:
ok thank you.
I tested my tricoptero Saturday 27/04/2019.
the tricoptero flew very well the original PID parameters worked very well, I had to make adjustments only in the PID, Roll, Pitch, Yaw.
Would you have any suggestions on how to put the location buzzer,
Thank you.
If someone wants to mount a tricopter like my follows photos and construction files I made it in the 3D printer.