iNavFlight/inav #4651: PPM input non functional for NOX target.

SeanTheITGuy, 2019-04-27:
## Current Behavior

Flysky i6x transmitter and FS-IA10B receiver hooked to [BG Nox Clone](https://www.banggood.com/20x20mm-Betaflight-F4-Noxe-MPU6000-Flight-Controller-AIO-OSD-5V-BEC-Built-in-LC-Filter-for-RC-Drone-p-1310419.html) shows no activity in the receiver tab.  It is not a wiring issue, because it works perfectly in BetaFlight.  It is possible that it is a configuration issue that I cannot seem to resolve.

## Steps to Reproduce

1. Installed latest ( INAV/NOX 2.1.0 Feb 25 2019 / 17:06:12 (65b0ec142)) iNav
2. Configured for PPM RX
3. No activity in receiver tab.

## Expected behavior

Accurate PPM RX data.

## Suggested solution(s)

## Additional context

https://pastebin.com/t8KtfZhQ

---

Board: https://www.banggood.com/20x20mm-Betaflight-F4-Noxe-MPU6000-Flight-Controller-AIO-OSD-5V-BEC-Built-in-LC-Filter-for-RC-Drone-p-1310419.html

- FC Board name and vendor:
- INAV version string:

# INAV/NOX 2.1.0 Feb 25 2019 / 17:06:12 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

shellixyz, 2019-04-27:
Can you confirm the PPM pad is connected to MCU's PB10 pin ?

SeanTheITGuy, 2019-04-27:
Not without a long tear down, which I am willing to do, but will not get to for a couple days.  Just entering this so you will not think I have abandoned thread.  I can confirm that out of the box, NOX target Betaflight 4.0.0 works properly with no wiring changes, if that is of any use.

shellixyz, 2019-04-27:
Well it is PB10 in Betaflight's target so it is a bug. Also uses the same timers. Did you have LED_STRIP enabled while testing ? Please test with this build of 2.2 beta to see if it solves the issue so we don't waste time trying to fix an issue that has already been solved: [inav_2.2.0_NOX.hex.zip](https://github.com/iNavFlight/inav/files/3124240/inav_2.2.0_NOX.hex.zip)

SeanTheITGuy, 2019-04-27:
Yes, it is working in 2.2 beta.  Thanks for your help.

closed 2019-04-27 by SeanTheITGuy

SeanTheITGuy, 2019-04-27:
One final thing. Enabling LED_STRIP does return the system to it's former non-working state, even in 2.2Beta.  Something there is the issue.  But I don't need that for now.