iNavFlight/inav #4654: IBUS Telemetry: Temperature displays wrongly

georgekucher, 2019-04-27:
## Current Behavior

After updating to version 2.1, FC temperature displayed on my transmitter shows from ~3.1 to ~4.3. It wasn't so before, when I was on 2.0.
[image]

## Steps to Reproduce

1. Upgrade to 2.1
2. Enable IBUS telemetry.
3.
4.

## Expected behavior

Temperature shows reasonable values. 
## Suggested solution(s)

Fix this issue)
## Additional context

[Dump](https://pastebin.com/VW3zqf43)
---

- FC Board name and vendor: INAV/MATEKF722 2.1.0 Mar 24 2019 / 22:52:35 (65b0ec142)
- INAV version string: GCC-7.1.0

shellixyz, 2019-04-27:
This is a know issue. It has been fixed already in 2.2 dev. Thanks for the report

closed 2019-04-27 by shellixyz

georgekucher, 2019-04-27:
Thanks.
Could you please provide a link to it?

shellixyz, 2019-04-27:
A link to a build of 2.2 dev or a link to the PR of the fix ?

georgekucher, 2019-04-27:
To the reported bug

georgekucher, 2019-04-27:
Or there wasn't?

shellixyz, 2019-04-27:
It is #4504

georgekucher, 2019-04-27:
ohh. I was looking in issues. thanks.
btw which temperature (what unit) is sent to ibus?

shellixyz, 2019-04-27:
It is decidegrees centigrade (with an offset because the radio expects an unsigned integer)

georgekucher, 2019-04-27:
no, I mean is it imu or baro or any else temperature))

shellixyz, 2019-04-27:
Baro temperature if there is one or gyro temperature otherwise