iNavFlight/inav #4655: Setting trims for Angle mode

MartinHugh, 2019-04-28:
Although I have gone through the whole mechanical trim itterations, followed by servo autotrim, I still find that my foam wing often needs roll trim applying when I do the first flight of the day.  This is not uncommon for me - with or without a FC.  I put this down to a 1 hour drive to the flying field where the wings expand and contract with the weather.

So,  Back to iNav.  Having adding in some roll trim so that it flies nicely in manual (and not redoing autotrim) I notice I now have a gentle bank in Angle mode.  I assume it sees the trim as stick input.

Perhaps I am approaching this from the wrong angle, and if so I would be grateful for some advice on how to deal with this.  I really dont want to be doing autotrim procedures every time I fly.

I note how two different FC systems deal with this - which seems much easier to do (unless I am missing a trick - which I could be).  The A3 stabiliser, reads the TX sticks at startup and uses that as the neutral (no input) setting.  The EagleTree Vector, allows trims to be "incorporated" into the FC really easily from a menu whilst flying.

Is there anyway this issue can be simplified to improve iNav, or alternatively if I am missing something, what is it.

Thank you for reading.

shellixyz, 2019-04-28:
Unless you fly in manual it is not a big deal if your control surfaces are not trimmed perfectly. If you trimmed in manual with the radio just go to the motors tab while in manual mode then note the servo positions and set that as the servo mid in the servo tab then reset the trims on your radio. But yeah this could be automated with an OSD menu item and a button in the GUI to automatically set the servo mids to the current position. Neutral position is always 1500 in all flight modes so if you trim through the radio then switch out of manual the aircraft won't fly as expected.

MartinHugh, 2019-04-28:
@shellixyz thank you for your comments.
I tend to use angle for launching (and sometime for landing if it is gusty) but generally like to fly in manual.

It would be nice if iNav could tolerate some trim on the radio and have the means to tell the FC that this is not stick input. 1500us as neutral seems a little restrictive IMHO.

Jetrell, 2019-04-29:
> 
> It would be nice if iNav could tolerate some trim on the radio and have the means to tell the FC that this is not stick input. 1500us as neutral seems a little restrictive IMHO.

Something like the ET Vector has. 'Incorporate radio trims"

MartinHugh, 2019-04-30:
Is there anyway we can look at adding a feature immediately sets  a new neutral position after togglng a switch on the tx whilst in the air ?

shellixyz, 2019-04-30:
Neutral for everything but manual ?

MartinHugh, 2019-04-30:
@shellixyz - manual should obviously be manually trimmed - and it is this which causes the issue for FC flight modes.  So yes neutral for eveything except manual.

digitalentity, 2019-04-30:
INAV doesn't care about trim values, it also doesn't have such thing as
stick center.
It expects zero values coming from the radio in all modes. Period. If you
want to shift stick zero in some modes - you'll probably have to use some
hackery on the radio.

MartinHugh, 2019-04-30:
@digitalentity thanks for your reply.

If iNav does not care about trim values, why after adding roll trim in manual does the aircraft then fly with a banked attitude in ANGLE mode.  I assumed that it simply seeing the trim as stick input ?

digitalentity, 2019-04-30:
@MartinHugh trim **IS** stick input. Your radio sends and INAV receives only a channel value.

MartinHugh, 2019-04-30:
@digitalentity that is my point.  The FC cannot distinguish between a trimmed channel or an untrimmed channel - that is why it is necessary to allow the pilot to inform the fc what is a hands off, neutral input.  Otherwise in (for example) angle mode, after trimming in manual, the aircraft suddenly stops flying straight and level.

This is not a specific iNav problem, every stabilsed FC has it, but many have instant simple remedies.  All I am trying to suggest is that one is added for iNav.

MartinHugh, 2019-05-02:
OK  chaps, so, any chance of anyone taking a look at this?

shellixyz, 2019-05-02:
We can add an option to transfer radio trims to servo mids easily in the GUI and CMS. Getting the stick center at boot-up to set servos mids doesn't sound like a great idea to me.

MartinHugh, 2019-05-02:
@shellixyz thank you for this.

closed 2023-10-27 by DzikuVx