iNavFlight/inav #4656: Servo output issue in OmnibusF4 V6

kandesigner, 2019-04-28:
Greetings all,

I can't figure out why servo outputs are not assigned in the iNav configurator 2.1.4 or older versions for my Airbot Omnibus F4 V6, flashed with:

INAV/OMNIBUSF4V6 2.1.0 Feb 28 2019

Screenshot below shows how the motors are assigned in strange way, however servos never shown as outputs, although servos are reacting visually in the Motors tab.

Note that the motors sequence and rotations are correct in the Motors tab:
PWM1 -> Motor 1
PWM2 -> Motor 2
PWM3 -> Motor 3
PWM4 -> Motor 4
PWM5 -> ...
PWM6 -> ...

[image]

[image]

Why Servos are not registered as outputs in the mixer tab? Thank you in advance for your generous support

kandesigner, 2019-04-29:
Oh.. that's very disappointing. I used Omnibus F4 V3 before and it works great as a quadcopter with servos, until got fried without a reason. Later I realized that many people complain about its reliability.

Do you suggest I wait for an updated firmware (assuming its not a hardware limitation) or I just pick another board?

Many thanks for your valuable feedback.

DzikuVx, 2019-04-29:
Pick another board. Pin allocation is at least strange and it seems like the hardware will not allow to use 2 last pins as servos in multirotor configuration

closed 2019-04-29 by DzikuVx

Cro2230, 2019-08-19:
Hello,
Hoping for some advice, if I was to make a custom firmware for a multirotor, would the below changes work in theory: 

Remove / Comment out 2 lines which use TIM8 for FW_MOTOR:
DEF_TIM(TIM8, CH2N, PB0,                    TIM_USE_FW_MOTOR,   1, 1), // S1_OUT    D(2,2,0),D(2,3,7)
DEF_TIM(TIM8, CH3N, PB1,                    TIM_USE_FW_MOTOR,   1, 1), // S2_OUT    D(2,2,0),D(2,4,7)

Change S5 and S6 from TIM3 to TIM8:
DEF_TIM(TIM8, CH3, PC8, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO,    0, 0), // S5_OUT    D(1,7)
DEF_TIM(TIM8, CH4, PC9, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO,    0, 0), // S6_OUT    D(1,8)

Also, the FC has CAM_C pin but is not defined anywhere. Is this something I can also add to the target to see if it works

Thanks in advance for any feedback.