iNavFlight/inav #4658: It's possibile to spin the motor without having to arm the board, maybe using an OSD command?

marcogarzotti, 2019-04-29:
Sometimes you just want to spin a motor but maybe you don't have a GPS lock because you are in your garage/basement testing or repairing some stuff, therefore you can't just arm and increase throttle.
In these conditions I usually have to remove the GPS feature and only then arm the FC without GPS lock. I have to admit that sometimes could be helpful to being able to spin a motor without messing with FW configuration, especially because a PC would be required. Everything went fine so far but the day i'll forget to re-enable something i'll surely swear a lot...
Wouldn't be useful yet still safe enough to have an OSD menù that can arm the board and/or let the motor spin **if specifically required and allowed from the pilot?** No matter what, GPS and other safety checks...
Maybe the function could be temporary (30sec or FW customizable) and signaled by a pesky buzzer noise, maybe it could requires the two separate confirmations by the pilot to being engaged, whatever it makes it reasonably safe.
Is it possible?
It's a no-no feature for some reasons that i don't get?
Thank you.

giacomo892, 2019-04-29:
In 2.2 it will be possible to bypass arming checks by flipping the ARM switch multiple times so you will be able to ARM and test the motors at your risk.

shellixyz, 2019-04-29:
@marcogarzotti You don't have to remove the GPS to be able to arm inside, there is the `nav_extra_arming_safety` setting for that. Just set it to off and you will be able to arm without a GPS fix.

closed 2019-04-29 by shellixyz

marcogarzotti, 2019-04-30:
@shellixyz: thanks, i knew that setting and/but in the end it's the same issue, you have to remove "hardcoded" safety feature for a temporary situation. *hardcoded = you need a pc and to remember to re-enable things.
@giacomo892 : thanks, that would be perfectly fine imho. Looking forward 2.2.