iNavFlight/inav #4661: Start motors when POSHold is active

Ravid824, 2019-04-30:
What is the reason that the motors will not start while the drone in PODHold mode?
If there is a good reason, why this option is not disabled temporary until the drone launch and then enabled automatically if the remote set to be in this mode.

DR

teckel12, 2019-04-30:
@Ravid824 Seems fairly obvious why this wouldn't be valid. I can see arming and the drone jumping several feet as the air pressure fluctuates. Or flying towards the "pilot". 

Keep in mind that with INAV the altitude is typically only controlled by the barometer. It's not like with the Mavic Pro which uses multiple optical flow sensors for position hold.

Ravid824, 2019-04-30:
It is clear, perfectly make sense. thank you
However, as I mentioned earlier, even so the user enable the POSHold, it should not be activated until the drone gain some height (lets say one meter above ground). At this point the software should activate the POSHold (if enabled by the user). 
We should eliminate as much as possible any user destruction or extra steps to do while flying.
The DJI controllers has similar mode, user can prepare and activate it before the flight (I am sure that they also relay on BARO reading). 

Thank you
DR

digitalentity, 2019-04-30:
"Gain some height" is the problem here.
We don't have "takeoff detector" sensor. Barometer tend to drift and you
will end up with poshold either not engaging (if baro drift is negative) or
engaging while still on the ground (if drift is positive).