iNavFlight/inav #4663: Adding waypoints in flight

michaelvanderhoek, 2019-05-01:
## Current Behavior
Current behavior is not available. Its a new feature.

## Desired Behavior
Making a waypoint mission on a tablet with an app like Mission Planner is a lot of fun, but sometimes beeing at the location gives you new insight in how to make a new mission. This would be especially a nice feature for filming with a drone. this way you can determine your filming locations and then execute the whole waypoint mission.

## Suggested Solution
When at the flying site, manouver the drone at the desired location, set poshold and flick a switch to set waypoint. Then continue to the next point, set waypoint and so on. Waypoints have to be stored in flight. 
When landed and disarmed waypointroute has to be saved to eeprom and set as current mission, disarming means mission had ended and can be saved. When taking off again, waypoint function executes the mission. When in poshold and the switch "set waypoint" is flicked again the previous route is deleted and all starts over again. 

If this all is possible, maybe it would be nice if the new route can be downloaded to an app to store it in a file for later flights

## Who does this impact? Who is this for?
All users

peteoz, 2019-05-05:
that's a great feature request actually. similar to how the old 3DR Solo used to work.
also considering the fact that without internet connection it's nearly impossible to create missions, this would give us new opportunity. I travel a lot in remote areas without internet for many days, how cool would it be to create new missions the way OP suggest.

michaelvanderhoek, 2019-05-05:
@peteoz, i didnt even think of that option! Thanks for adding!

b14ckyy, 2021-05-11:
Agree, sounds like a good idea. 
@stronnag also has released a tool to transform blackbox log pathes into waypoint missions. This might also be useful to do something similar but manual waypoints would be great too.

closed 2021-11-07 by breadoven