iNavFlight/inav #4664: Traditional helicopter support

vltsvn, 2019-05-01:
Hi, guys!
I want to use INAV on a traditional helicopter. I have read the source code and there is helicopter support described there. But the docs say that only multicopters and planes are supported. I think, adding helicopter support will not be too hard.
I see it like:
1. Add "collective pitch" feature
2. Add some functionality to calculate throttle from collective pitch (pitch curves or maybe gouverneur).
3. Add inav-configurator support.

I have some helicopters of different sizes and some Matek controllers and ready to test INAV on helicopter.

Jetrell, 2019-05-03:
I have two 600 machines from back in the day when I used to fly 3D smack.  I have often thought it would be nice to have a FC onboard, to fly one with FPV.  But I have never found software that supports cyclic and collective control, with RTH ability.

@vltsvn  I second your request.

vltsvn, 2019-05-03:
I have written a letter to digitalentity, the creator of INAV. He said, that  the team is really interested in helicopter support, but it is not a priority task. I will try to adapt INAV to traditional helicopter by myself. I have a lot experience in STM32 programming and work in airspace industry. I will test the code on my helicopters and write results here.

KenImhof, 2019-05-11:
You may want to try a 3d quad that will fly just like a 3d helicopter but much better. I have already modified iNAV to fully support 3d mode. I have also added many features such as; two hardback levels, rescue, inverted navigation position hold for learning inverted 3d controls, automatic rth when battery is low, and many other features. 

Software and detail are available at:
https://github.com/KenImhof/iNav3d/releases

Please ignore the videos as they taken when this project first started and have not been updated. There are many of us 3d helicopter pilots running this software and it is working great and is a great training aide / simulator for 3d pilots.

Jetrell, 2019-05-13:
> 
> 
> You may want to try a 3d quad that will fly just like a 3d helicopter but much better. 

Better?.... With all respect... I don't think so.

I've flown 3D helis for a decade. And have also used BF with 3D quad mode. 
The authority of collective pitch on a 3D heli is far superior in response, than changing the direction of a motor and propeller with a quadcopter.   
Even simple moves like a Tic Toc are so laggy with motor reversal in 3d quads, compared to collective pitch control on a 3D heli.

vltsvn, 2019-06-11:
@Jetrell, I'm allmost ready with my iNav port for traditional helis. It's pretty ready to fly test. I'm finishing some minor fixes and writing setup docs for now. I hope, I'll start fly tests in a week or so. From your message above I see you are an experienced RC heli pilot. I'm sure you can correctly setup and configure an rc-model. Would you like to join testing of helicopter support in iNav before these changes would be included in this repo (I hope, in 2.3.0). The firmware doesn't support 3D flight for now, but it supports cyclic and collective pitch, CCPM120, pitch to throttle curve, external guverneur, It should support allmost all flight modes of iNav includding RTH. If you are interested in testing this firmware, contact me on Telegram (Vitaly Vyaltsev), I'm in INAV Telegram channel.

Jetrell, 2019-06-12:
@vltsvn Thank you for the offer. But at this time I don't have all the hardware I would need to test it out for you. 
Being a larger heli, with high current brushless servo's. I will need a servo power distribution board. And also have to find another more suitable FC.
I'm even weighing up making a specialized 450 size first, later on.

Thank you for your efforts and time in writing this code. Keep us informed of your progress.

myheli, 2019-08-17:
Hi, wanted to ask, how is the state of things and whether traditional helicopter now works?

iggut, 2019-08-19:
Also interested in the progress of this. I am a small Align 250 waiting for it! Thank you for your work!!

MrLatebros, 2019-10-08:
@vltsvn I've sent you a message on telegram if you want me to test it.

lucag, 2019-12-13:
@vltsvn I'm interested too. I'd love to try your code and willing to contribute some.

NeilGaede, 2020-01-25:
Vltsvn do you have an e-mail address? Or are we not allowed to give them out. Can you post the firmware? I have a HoneyBee King 2 I am setting up for FPV and I already have the flight controller wired up and ready to go. All I need now is the code.

KenImhof, 2020-01-27:
@Jetrell,

Here is one of my flights from the field today.  It is a micro quad for around 100USD and fly's better than any of the micro helis.  I also have some 10-Inch quads that fly even better.  Big advantages are the repair costs and I typically use them as a simulator for my big helis.  These are nearly indestructible and cost is a fraction of a heli.

https://www.youtube.com/watch?v=nlYEZhBlHyQ

kaliblazin707, 2020-02-28:
Hey, I just picked up one of these: https://m.banggood.com/OMPHOBBY-M2-6CH-3D-Flybarless-Dual-Brushless-Motor-Direct-Drive-RC-Helicopter-BNF-With-4-IN-1-Flight-Controller-p-1576058.html?akmClientCountry=America&utm_source=Youtube&utm_medium=cussku&utm_campaign=essentialrc&utm_content=zhikejian&p=R427033382918201512W&ID=6276384&cur_warehouse=CN

Would love to have some sweet sweet inav goodness

mikeo0, 2021-03-20:
> I have two 600 machines from back in the day when I used to fly 3D smack. I have often thought it would be nice to have a FC onboard, to fly one with FPV. But I have never found software that supports cyclic and collective control, with RTH ability.
> 
> @vltsvn I second your request.

Arducopter will do it

jasc76, 2021-08-18:
Any progress on this? As I come from BF and INAV and all my gear is tailored towards this kind of stuff I find it intriguing to have an fc with inav doing this whole FBL thingi and having the sensors at my OpenTX radio. Not so much for RTH, more to have a known ecosystem to deal with?

Engineereric, 2022-02-09:
Old probably version 1 or 2 of BF had the mixing for heli. 
Cant some one just put this back in so that we dont have to do a custom mix. 
It is trivial and I dont understand why it has not been done.

MrD-RC, 2022-02-10:
@Engineereric a tiny percentage of people use iNav for helis. To you it may seem trivial. But I wouldn’t even know what is needed for a heli mix. 

Maybe if you share a heli mix, and some well tested default settings for a heli, something could be done about it. 

Also, iNav is not BetaFlight. It is a form from CleanFlight.

closed 2024-02-02 by DzikuVx