iNavFlight/inav #4665: slew/stick position pan speed for camera

P-I-Engineer, 2019-05-01:
## Current Behavior
doens't exist

## Desired Behavior
use a stick to position to determine the speed to rotate a servo. When stick centered, servo will stay at last position. 

## Suggested Solution
institute this method straight into inav
https://openrcforums.com/forum/viewtopic.php?t=856

I'd prefer to see this on a switch, 
switch positions:
low, servo at center 1500us position
high, pan stick is active and other functions of stick are suppressed.
rudder stick on a wing makes sense
when switch back to low, camera returns to center position.

## Who does this impact? Who is this for?
airplane users, mysefl

## Additional context
Obviously i can do this in opentx, and have. I really like the feature. I thought it'd might be interesting to non opentx users as a built in function.
I've used this in the past as a way to control pitch on a gimbal with a head tracker. It allowed one man filming.

thanks

DzikuVx, 2019-10-25:
OK, I will try to put it on my todo list

DzikuVx, 2020-05-27:
```
gvar 0 0 -500 500
logic 0 1 -1 15 1 7 0 1500 0
logic 1 1 -1 17 4 0 0 10 0
logic 2 1 -1 19 0 0 4 1 0
logic 3 0 -1 0 0 0 0 0 0
logic 4 0 3 0 0 0 0 0 0
logic 5 0 1 0 0 0 0 0 0
logic 6 0 2 0 0 0 0 0 0
logic 7 0 5 0 0 0 0 0 0
logic 8 0 -1 0 0 0 0 0 0
logic 9 0 -1 0 0 0 0 0 0
logic 10 0 -1 0 0 0 0 0 0
logic 11 0 -1 0 0 0 0 0 0
logic 12 0 -1 0 0 0 0 0 0
logic 13 0 -1 0 0 0 0 0 0
logic 14 0 -1 0 0 0 0 0 0
logic 15 0 -1 0 0 0 0 0 0
```

The snippet above allows to use Logic Conditions and Variables in INAV 2.5 to build slew rate on a RC channel. In the example above:

* RC channel 7 is assigned to a POT
* GVAR0 will increase when RC channel 7 will be above 1500
* GVAR0 will decrease when RC channel 7 will be below 1500

what is left is to assign GVAR0 into one of servo mix rules.

GUI is provided by Configurator 2.5 in Programming Tab

closed 2020-05-27 by DzikuVx