iNavFlight/inav #4666: OSD artificial horizon improvement & instant consumption alarm

FOX1185, 2019-05-02:
Goodmorning everyone; in my humble opinion it could be useful to insert more lines in the artificial horizon in order to have a better information on the inclination. Unfortunately the single line of the horizon disappears from the screen beyond the approximately 20 ° and therefore in the steep climbs or descents when for example you pass in the clouds you tend to lose the information on the inclination; in the image below a draft of how to increase the lines; a really nice thing would be to implement something like the real hud with 
lines of equal length with side readings in degrees but I don't know if the OSD software allows it.
It could also be useful to have a flashing alarm on the instantaneous consumption which can be set (as for the timer, altitude and distance) so as to be able to better manage the liion batteries which have reduced possibility of discharge.

Do you think these things can be useful to implement?
Thank you and greetings.

[image]

[image]

Jetrell, 2019-05-05:
@FOX1185  At present iNav already has flashing alarms that can be set for -  altitude, distance, time and low battery capacity.
What you are requesting for the AHI is mostly beyond the ability of a character based OSD, which most open source FC use, due to the hardware that is supplied on the FC boards.
Although, one of the developers @fiam has started the early stages of pixel based OSD. 
So what you are requesting may occur down the road.

Jetrell, 2019-05-05:
@FOX1185  At present iNav already has flashing alarms that can be set for -  altitude, distance, time and low battery capacity.
What you are requesting for the AHI is mostly beyond the ability of a character based OSD, which most open source FC use, due to the hardware that is supplied on the FC boards.
Although, one of the developers @fiam has started the early stages of a pixel based OSD. 
So what you are requesting may occur down the road.

shellixyz, 2019-05-05:
@FOX1185 The default is 20° max but you can set it to the value you like with `osd_artificial_horizon_max_pitch`

ShawnzAZ, 2019-05-28:
You can do this with the Radix FC (Vector Graphics OSD) and their INav fork, FWIW.   I did not get the artificial horizon in 2.1, but in 2.0.1 it works fine.  It looks similar to the Eagletree Vector's horizon bars, only not in color :).

closed 2019-08-10 by stale[bot]