iNavFlight/inav #4668: GPS Timeouts after 2.0.1+ version

VBKesha, 2019-05-02:
Hello!
I try to update on version 2.1.0 from 2.0.0 and after it all work but GPS.
I use "BN-220 NEO-M8N GPS" and Omnibus F4 V3. Module configured to use  NMEA protocol.
And after update every 3-5 second i have beepeng and increase GPS timeout errors.
I try to downgrade to 2.0.1 version and error not fix.
After i downgrade to 2.0.0 all work fine again.

giacomo892, 2019-05-02:
Configure the port at 115200bps and set the GPS protocol to UBLOX7.
Then post a _diff_ and report back.

MartinHugh, 2019-05-03:
Anecdotal only I know.  But I use this GPS unit with Matek F411-Wing and had the same issue on 2.1 two weeks ago, although I never tried it in 2.0

My original setting were
baud rate = 38400
gps_provider = NMEA
gps_sbas_mode = AUTO

I changed the settings to 

baud rate = 115200
gps_provider = UBLOX7
gps_sbas_mode = EGNOS

Worked fine then.  I flew with 21 sats and HDOP <1.0

VBKesha, 2019-05-03:
On UBLOX7 working fine.
But i try to repeat bug with NMEA and cant got error 1-2 hour. All working fine. But when i open window and put fly conroller outwindow the bug repeated. In apartment i have temperature at 24-26C, at street 10-12C.
I think its frequency drift due to temperature, on this FC a have strange ESC behavior(engine have some rpm then throttle set to 0) when temperature less 4C, but I was thinking on the ESC.

giacomo892, 2019-05-03:
NMEA shouldn't be used when UBX is supported.

VBKesha, 2019-05-04:
Yes, but i think NMEA need do be fix too.

EnsignCrasher, 2019-06-13:
same issue here on 2.1.0
I thought about downgrading to 2.0.0, but I'm rather afraid of the "GPS lag" bug....

d2f2, 2019-11-23:
On 2.3.0-RC2 with sp racing f3 nmea works ok, until enable fport. With fport enabled there is many timeouts and gps faulty.

cochrane771, 2020-03-31:
I have this problem also. any fixes? I need to use nmea gps to work with mfd crossbow mini antenna tracker. tracker does not seem to work with ublox gps.

OMGChase, 2020-07-04:
Same issues happening on my side.
Every minute or so, getting timeouts and errors on GPS statistics.

Omnibus F4 V3
Beitian 220T

NMEA Protocol
115200 Baud rate

covert-, 2022-09-25:
This seems to still be a bug in 5.1.0 with processing NMEA 

I have multiple GPS modules with NMEA only (no UBLOX) and tried them on 2 different UART ports with totally different wiring and they all exhibit similar behavior. GPS icon goes blue, Total messages count increase every second (they're 1hz) . Then between 10-40 seconds the GPS icon goes red and total messages count stops increasing. Take 10 or so seconds for GPS icon to go blue again and message count to start to increase. Checked with gpspassthrough and everything looked stable in u-center (NMEA mode) with NMEA message sets every 1 seconds.

Also flashed 4.0.0 just to check. exact same problem.

Multiple baud rates from 9600 to 115200 tried all exhibiting the same behavior.

I dug an old ublox LEA-6H from the parts bin and wired it into one of the ports and it worked perfectly with no red GPS dropout like the other ones,

I currently have both NMEA and UBLOX and I can switch from one to the other and replicated the bug consistently.

Board F4 Omnibus pro

stronnag, 2022-09-25:
This would appear to be a  connection or possibly a configurator issue  rather than a firmware issue.

A BN880 in NMEA mode works perfectly with mwp as the visualisation tool (over 30 minutes so far with no GPS drop outs).

covert-, 2022-09-26:
I tried with 6.0.0 and still the same issue.

Configuration is blank from full erase flash except for setting up the port and GPS.

I've double checked connection and wired in a second GPS with new wiring just to confirm it's not a connection issue. It's also on a physically different plug on the board. Results are consistent on both ports with different wiring.

I'll can check the signal with a oscilloscope later on but I doubt the results will change.

I'll also try another F4 board.

MartinHugh, 2022-09-26:
> I tried with 6.0.0 and still the same issue.
> 
> Configuration is blank from full erase flash except for setting up the port and GPS.
> 
> I've double checked connection and wired in a second GPS with new wiring just to confirm it's not a connection issue. It's also on a physically different plug on the board. Results are consistent on both ports with different wiring.
> 
> Increasing the port speed from 9600 to 115200 doesn't increase the amount of times the error occurs. If it was a connection issue you would expect errors to increase with the data speed.
> 
> I'll can check the signal with a oscilloscope later on but I doubt the results will change.
> 
> I'll also try another F4 board.

Are all your GPS cables properly screened.  This can make a night and day difference if there is noise nearby.

stronnag, 2022-09-26:
Wait a second. The BN880 does 5Hz in NMEA mode, so it won't fail a 1Hz test. 
Let's find the culprit ....

```
// GPS timeout for wrong baud rate/disconnection/etc in milliseconds (default 2000 ms)
#define GPS_TIMEOUT             (1000)
```
Yeah right ... Let me fix this.

covert-, 2022-09-26:
> Wait a second. The BN880 does 5Hz in NMEA mode, so it won't fail a 1Hz test. Let's find the culprit ....
> 
> ```
> // GPS timeout for wrong baud rate/disconnection/etc in milliseconds (default 2000 ms)
> #define GPS_TIMEOUT             (1000)
> ```
> 
> Yeah right ... Let me fix this.

Thanks for finding that. I knew I was not going crazy.

stronnag, 2022-09-27:
Fixed by #8423

closed 2022-09-27 by stronnag