iNavFlight/inav #4669: Tons of new setting default ranges that are preventing model arming after update

simpert, 2019-05-03:
## Current Behavior

## Steps to Reproduce

1. Update inav to latest
2. use cli to load all current model setting that have working for a year
3. observe all the pre-arm check failures by reading status in console observing failed value and updated it to some new range then repeat over and over untill all new defaults are within range
4.

## Expected behavior

After update my model settings and profile should still work
## Suggested solution(s)

Put some notes somehere obvious of each new default. I had my auto nav RTH settings very slow and that how I liked it. The update is forcing me to increase all values to these new range defaults
examples:
nav_mc_braking_speed_threshold
nav_mc_braking_boost_speed_threshold
nav_mc_braking_bank_angle
nav_fw_bank_angle
nav_fw_climb_angle
nav_fw_pitch2thr
nav_fw_launch_accel
nav_fw_launch_detect_time
nav_fw_launch_idle_thr

even Fixed Wing Navigation Settings were preventing me from arming

CMON, took about 20 minutes to go through and set all these non-needed settings to allow me to arm again after the update

## Additional context

---

# version
# INAV/MATEKF722 2.1.0 Feb 25 2019 / 16:59:54 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

- FC Board name and vendor:
MATEKF722
- INAV version string:
2.1.0

> Heres my `diff`
`# resources

# mixer
mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mix

# servo

# feature
feature -AIRMODE
feature THR_VBAT_COMP
feature GPS
feature LED_STRIP
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper

# map
map TAER

# serial
serial 0 0 115200 38400 0 115200
serial 2 32 115200 38400 0 115200
serial 3 2 115200 38400 0 115200
serial 4 2048 115200 38400 0 115200

# led
led 0 7,7:N:G:0

# color

# mode_color
mode_color 6 7 1

# aux
aux 0 0 0 1500 2100
aux 1 1 1 900 1125
aux 2 2 1 1325 1650
aux 3 20 0 1475 2100
aux 4 20 2 1450 2100
aux 5 3 6 1400 2100
aux 6 9 6 1775 2100
aux 7 8 7 1825 2100
aux 8 12 4 975 1025
aux 9 17 5 1375 1600
aux 10 32 3 1275 1525
aux 11 33 3 1550 1825
aux 12 34 3 1850 2100

# adjrange

# rxrange

# temp_sensor

# osd_layout
osd_layout 0 0 26 0 V
osd_layout 0 1 1 0 V
osd_layout 0 2 0 0 V
osd_layout 0 3 0 0 V
osd_layout 0 4 0 0 V
osd_layout 0 7 13 10 V
osd_layout 0 8 9 12 V
osd_layout 0 9 0 6 V
osd_layout 0 10 11 0 V
osd_layout 0 11 1 3 H
osd_layout 0 12 1 4 H
osd_layout 0 14 25 1 V
osd_layout 0 15 12 6 V
osd_layout 0 22 14 4 V
osd_layout 0 28 23 11 H
osd_layout 0 29 24 15 H
osd_layout 0 30 1 11 V
osd_layout 0 32 1 1 V
osd_layout 0 33 0 7 V
osd_layout 0 34 0 0 H
osd_layout 0 38 8 1 H
osd_layout 1 0 26 0 V
osd_layout 1 1 1 0 V
osd_layout 1 2 0 0 V
osd_layout 1 3 8 6 V
osd_layout 1 4 8 6 V
osd_layout 1 7 12 4 V
osd_layout 1 9 24 5 V
osd_layout 1 10 11 0 V
osd_layout 1 11 0 3 H
osd_layout 1 12 0 4 H
osd_layout 1 14 25 1 V
osd_layout 1 15 25 2 V
osd_layout 1 19 0 6 H
osd_layout 1 23 12 9 V
osd_layout 1 28 17 1 H
osd_layout 1 30 1 13 V
osd_layout 1 32 1 1 V
osd_layout 1 33 24 4 V
osd_layout 1 34 10 2 V
osd_layout 1 36 0 5 H
osd_layout 1 38 0 2 H
osd_layout 2 0 27 0 V
osd_layout 2 1 0 1 V
osd_layout 2 7 12 1 V
osd_layout 2 8 8 0 V
osd_layout 2 9 2 8 V
osd_layout 2 15 2 9 V
osd_layout 2 29 24 1 V
osd_layout 2 32 0 2 V
osd_layout 2 34 11 15 V
osd_layout 2 38 0 0 V
osd_layout 2 48 24 15 V
osd_layout 2 49 25 14 V
osd_layout 2 53 12 2 H
osd_layout 3 0 26 0 V
osd_layout 3 10 11 0 H
osd_layout 3 16 1 0 V
osd_layout 3 17 1 1 V
osd_layout 3 18 1 2 V
osd_layout 3 56 1 3 V
osd_layout 3 57 1 4 V
osd_layout 3 58 1 5 V
osd_layout 3 59 1 6 V
osd_layout 3 60 1 7 V
osd_layout 3 61 1 8 V
osd_layout 3 79 4 9 V
osd_layout 3 80 4 10 V
osd_layout 3 81 4 11 V
osd_layout 3 84 4 12 V

# master
set looptime = 500
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 100
set gyro_notch1_hz = 200
set gyro_notch1_cutoff = 100
set acc_hardware = MPU6500
set acczero_x = -11
set acczero_y = -23
set acczero_z = -12
set accgain_y = 4092
set accgain_z = 4098
set align_mag = CW270FLIP
set mag_hardware = QMC5883
set magzero_x = -54
set magzero_y = 81
set magzero_z = 140
set baro_hardware = BMP280
set pitot_hardware = NONE
set blackbox_rate_denom = 2
set min_throttle = 1040
set max_throttle = 2000
set motor_pwm_rate = 2000
set motor_pwm_protocol = MULTISHOT
set failsafe_throttle = 1270
set failsafe_procedure = RTH
set model_preview_type = 3
set gps_sbas_mode = AUTO
set gps_min_sats = 5
set deadband = 2
set yaw_deadband = 3
set inav_w_z_baro_p =  0.450
set inav_w_z_gps_p =  0.250
set nav_auto_speed = 400
set nav_auto_climb_rate = 300
set nav_manual_speed = 400
set nav_landing_speed = 110
set nav_land_slowdown_minalt = 50
set nav_land_slowdown_maxalt = 501
set nav_emerg_landing_speed = 150
set nav_min_rth_distance = 5
set nav_rth_climb_first = OFF
set nav_rth_alt_mode = CURRENT
set nav_rth_altitude = 150
set nav_mc_bank_angle = 35
set nav_mc_hover_thr = 1290
set nav_mc_braking_speed_threshold = 1000
set nav_mc_braking_disengage_speed = 15
set nav_mc_braking_timeout = 1400
set nav_mc_braking_boost_factor = 0
set nav_mc_braking_boost_timeout = 1200
set nav_mc_braking_boost_speed_threshold = 1000
set nav_mc_braking_boost_disengage_speed = 10
set nav_mc_braking_bank_angle = 30
set nav_fw_cruise_thr = 1863
set nav_fw_min_thr = 1000
set nav_fw_bank_angle = 10
set nav_fw_climb_angle = 15
set nav_fw_dive_angle = 44
set nav_fw_pitch2thr = 100
set nav_fw_loiter_radius = 100
set nav_fw_cruise_speed = 40
set nav_fw_land_dive_angle = 0
set nav_fw_launch_velocity = 5000
set nav_fw_launch_accel = 1000
set nav_fw_launch_detect_time = 10
set osd_video_system = NTSC
set osd_units = IMPERIAL
set osd_rssi_alarm = 45
set osd_time_alarm = 7
set osd_alt_alarm = 12
set osd_dist_alarm = 805
set osd_neg_alt_alarm = 0
set osd_crosshairs_style = AIRCRAFT
set osd_left_sidebar_scroll = ALTITUDE
set osd_right_sidebar_scroll = HOME_DISTANCE
set osd_sidebar_scroll_arrows = ON
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5917

# profile
profile 1

# battery_profile
battery_profile 1

set vbat_min_cell_voltage = 340
set vbat_warning_cell_voltage = 380
set battery_capacity = 2200
set battery_capacity_warning = 660
set battery_capacity_critical = 440

# version
# INAV/MATEKF722 2.1.0 Feb 25 2019 / 16:59:54 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

`

giacomo892, 2019-05-03:
All you have to do is read the release notes from the next version you are running till the ones of the version you intend to run.
It will fix most of your issues.
And no, you cannot expect to paste a one year old configuration into the latest version and expect everything is working as before. Most things will work, some requires adjustments and some are deprecated.

simpert, 2019-05-11:
@giacomo892 I observed all release notes as far back as they go in repo and was unable to find any notes relating to new required min, max, or range settings. I realized this issue would best have been placed in the inav-configurator repo. 

I certainly do not believe this to be a `support` issue rather perhaps a feature request.

The user interface has but a few tooltips and they are incredible useful when trying to setup a new model and preventing the pre-arm check from disabling arming due to not providing a required value in the required range. For example many settings will not report any error and in ui do not have tool tips reporting valid ranges so the only way I was able to get these setting in range to allow pre-arm check pass was to hook up to terminal and see the last reported pre-arm check failure as it only shows one. then using terminal get the value of that property which shows required range there and set to valid value that settings. Then try to arm again. Rinse and repeat until determine valid ranges for all settings.
I believe providing tool tips or valid ranges for all required settings in the user interface of configurator would be a greatly beneficial feature.

I would be happy to code it myself and do a pull-request if someone is willing to point me in right direction in the ui project.

giacomo892, 2019-05-11:
You can find max/min here https://github.com/iNavFlight/inav/blob/master/src/main/fc/settings.yaml