iNavFlight/inav #4672: Ailerons don´t work in acro/manual but work in angle mode

pablorod14, 2019-05-03:
Hi, I have set up an airplane with inav, with a f722 wing fc. All is fine, but when I use acro mode or manual mode, ailerons don´t work, they just don´t move at all. There is no difference in behaviour between acro and manual in the ailerons, they just don´t move and don´t response. Rudder and elevator both work in acro and manual. But, in angle mode, all works fine. Ailerons move, rudder, elevator... I don´t understand what may be happening.

Dump:
https://pastebin.com/62N65Xx5

---
INAV/MATEKF722SE 2.1.0 Feb 25 2019 / 17:01:07 (65b0ec142)
GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

- FC Board name and vendor: Matek f722 WING

shellixyz, 2019-05-03:
When you switch to acro/manu are your roll channel moving in the receiver tab when you move the roll stick on your radio ?

pablorod14, 2019-05-04:
Yes, the mode in the osd and in the telemetry changes correctly, also in inav

shellixyz, 2019-05-04:
I am not talking about the mode I am talking about the roll channel. The only reason I can see for this issue is that your radio is programmed the override the roll channel or something when your mode switch is in the acro/manual position.

pablorod14, 2019-05-04:
Thanks! That was it,a silly mistake. Now all channels work. Just that I had to delete stabilized yaw on mixer because in acro/manual mode yaw barely moves, so in the mixer I had to select RC yaw so it works.

Thanks a lot, shellixyz.

closed 2019-05-04 by pablorod14