iNavFlight/inav #4674: F405 wing OSD lacks parameters

letruong6006, 2019-05-04:
I am using F405 Wing. INAV 2.1.4, has updated the latest software for Fc. when i enter the OSD section. I don't see the arrow indicating the direction of the home, the distance of the home, the GPS parameter .... what is the error? I need help
[image]
[image]
[image]
https://scontent.fhan4-1.fna.fbcdn.net/v/t1.15752-9/58968012_2343024319314286_6141478765680656384_n.png?_nc_cat=104&_nc_oc=AQk1bUJkRGgFRHUp4U_50ZDkFiYCC7nHwaqWnumHlokqro9LGP6Q_QaLMB1gQhNF77s&_nc_ht=scontent.fhan4-1.fna&oh=35df762bfc52cd12399870e2e4958505&oe=5D6A81D7
[image]

letruong6006, 2019-05-04:
[image]

letruong6006, 2019-05-04:
[image]

shellixyz, 2019-05-04:
[image]

letruong6006, 2019-05-07:
thank you shellixyz. I found them in OSD