iNavFlight/inav #4675: [SUPPORT] MAG3110 not detected

pfz4, 2019-05-04:
Hey,
I've bought me an external MAG3110 to cancel out the magnetic waves of my Power Distribution board [https://www.amazon.de/gp/product/B01NCSXWDN/](https://www.amazon.de/gp/product/B01NCSXWDN/). I've attached the Module to my F3 Flight Controller Deluxe [F3 Flight Controller Manual](http://seriouslypro.com/files/SPRacingF3-Manual-latest.pdf). My Arduino Nano detects the I2C Device with adress 0x0E. My FC doesn't detect the Magnetometer.

Thanks for any reply

shellixyz, 2019-05-04:
Make sure there are pull-up resistors on the I²C line of either you mag or your flight controller. Between 1k and 2.7k ohm.

pfz4, 2019-05-04:
I've redesigned my Copter and now I don't have use an external MAG. Thanks...

closed 2019-05-04 by pfz4

Jacky2k, 2020-01-04:
Hi,
the problem is not solved anyway. I think I found the problem and fixed it in https://github.com/iNavFlight/inav/pull/5309. Maybe that helps someone until the patch is upstream.