iNavFlight/inav #4682: Matek-f411 rssi firmware question

walney-reptiles, 2019-05-04:
Okay so please forgive me but i am new to the inav world,

heres a list of the stuff i have for my build

I have 
Matek F411 Wing Flight controller
Matek VTX-HV Video Transmitter
Runcam Micro Eagle
Deltas 8 Multi-protocol Receiver
ublox M8N GPS/Compass

im wanting to use Rssi on either my osd or Taranis but im not sure what firmware to use on the FC im guessing the right firmware is the Matek f411 Rssi but what i want to know is where i wire the GPS and compass, Receiver and rssi from channel 1 on the Receiver.

Anyhelp appriciated, thanks Josh.

Jetrell, 2019-05-05:
Off hand.... I think the Delta's 8 only has an SPPM output.  That will be an issue with the RSSI firmware. Both SPPM and RSSI require the ST1 pin for use.

While the GPS will use either Uart and the Compass will use the I²C SDA and SCL pins.

DzikuVx, 2019-05-09:
It's described in https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20Matek%20F411%20Wing.md

RSSI target allows to connect analot RSSI pin on ST1 pad of a flight controller.

closed 2019-05-09 by DzikuVx

Piotr-PB, 2019-10-26:
Hello,
I loaded F411 RSSI firmware and connected RSSI input to ST1. What do I set in iNAV configurator to make it work and display in OSD? Now it displays 0 which is not correct as receiver is getting signal.