iNavFlight/inav #4686: "Auto take off" for Quads

snakeeater12, 2019-05-06:
Hi Github

Does anybody know a CLI command or a mode for "Auto Take off" for Multicopters?

I need my Quad X once armed, to slowly take off and climbs to a height of one meter. Once reaching the one meter it locks (Altitude hold/ surface mode) without having to engage a seperate switch?

Thanks Snakeeater12

For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)

rb1205, 2019-05-06:
The altitude hold can do what you're asking, but it won't stop at a given altitude. Also only one meter of height is way too low: the barometer cannot guarantee that kind of precision, expecially over time, even moreso with position hold.

snakeeater12, 2019-05-07:
Thanks rb1205

I heard the same thing could be achieved using a sonar module (I heard they are more accurate at low altitudes) like the US-100, HC-SR04 or another rangefinder for example, could Altitude Hold mode or Surface mode work in conjunction together with the sonar module (to get the accuracy)?

Any ideas how to preset a height? (1 meter in my case) so I arm, then it takes off automatically and rises to the height, then I am able to fly in Angle mode, without any flicking of extra switches?

Thanks once again snakeeater12

teckel12, 2019-05-07:
@snakeeater12 INAV doesn't have the ability to do what you're asking. There's no settings that will do it.

DJI drones perform this with a combination of distance sensors, optical flow sensors and barometers. They're also working with known hardware with known flight characteristics.

Even with the correct sensors and programming to match, INAV will never have the known hardware flight characteristics like a commercial drone.

Not that's it's impossible, but it's a lot more complicated than it seems, especially because every quad has different flight characteristics.

Instead, arm in horizon mode, fly to about 9 feet, switch to position hold mode. It's not that hard.

snakeeater12, 2019-05-10:
Thanks teckel12 and rb1205 for helping

snakeeater12

closed 2019-05-10 by snakeeater12

Black6spdZ, 2019-11-06:
after the PIDs are tuned why would this be to diffficult to manage? I'm sorry but the open source FC firmware is falling WAY behind.. lets get some good sensors to use and make it happen. Are DJI coders that much more talented then the pool of contributors here?

rb1205, 2019-11-06:
DJI coders have the control of what hardware their code will run on, this firmware runs on everything from 2" micro quads to 30kg opta copter to helis.

Black6spdZ, 2019-11-07:
This is the excuse I hear from iPhone fanboys why they are so much better.. There is an auto takeoff flight mode in ArduPilot PX4 firmware..

digitalentity, 2019-11-10:
@Black6spdZ you really have only two options of getting this done fast:

1. Implement that in INAV yourself - the open-source way.
2. Pay some money to somebody to implement a specific feature for you - the outsource way.

Passive aggression will only get you banned for violation of our [code of conduct](https://github.com/iNavFlight/inav/blob/master/docs/policies/CODE_OF_CONDUCT.md).

Black6spdZ, 2019-11-10:
You call that passive aggression? I was only stating that comment of controlled hardware from rb1205 was the only reason DJI could make such functionality work when in fact this is already in another open-source FC firmware that can be installed onto any diy quad.