iNavFlight/inav #4688: Flexible GPS speed for Aircraft Navigation

robthomson, 2019-05-06:
If running a low speed gps - such as a cheap NMEA system, you get frequent issues with inav loosing sync with the gps - resulting in inav attempting to renegotiate gps comms.

When using systems like the arkbird tracker - piggy backed off the FC gps, this renegotiation plays havic with the tracker as during renegotiation all tracking fails.

It would be good to have a solution in place for 'WINGS' in which high accuracy navigation is not required.  This would allow use of cheap nmea bases gps modules on wings.

Essentially what is needed is an option in inav that is 'be happy with low speed gps and not throw errors'.  So.. assume gps is good!

This will allow trackers like skylark and arkbird to piggyback the gps signal within inav.

digitalentity, 2019-05-06:
INAV requires GPS updates at least at 5Hz rate. If you GPS can't cope with that - it's not compatible with INAV.

robthomson, 2019-05-06:
I understand that.

But reality is aircraft navigation does not need that refresh. 

Totaliy get is why you want that in a quad. 

Problem I have found is simple.  It is impossible to get that refresh rate on any nmea based GPS!

giacomo892, 2019-05-06:
>But reality is aircraft navigation does not need that refresh.
Why do you assume this, even more after after @digitalentity post?

If you need an NMEA GPS for the tracker you can install an additional BN180 on the machine, it's just light and very small.

robthomson, 2019-05-06:
I do just install a second gps.

It is not a matter of assuming a more accurate refresh rate is needed.

Historically flight controllers have managed quite fine in low refresh rate.  Yes.. Not perfect.. But nobody had an issue in the pre ublox days with their fc like. 

Ardupilot 
Cyclops storm
Arkbird 
Etc... 

All of these work fine with low speed nmea.   

As I stated.  Low speed is now good for high accuracy positioning for likes of quad copters.   But plenty fine for basic rth on wings.

digitalentity, 2019-05-08:
We feed GPS data along with other things into position estimation core. Navigation doesn't use the raw GPS data. Position estimation core requires a reasonable refresh rate to function properly.

I'm against tweaking position estimation code towards reducing estimation reliability and quality to support outdated GPS modules for previous century that can't manage 5Hz update data.

If we need a low speed/low bandwidth NMEA for the tracker - creating an NMEA telemetry output is a better solution.

robthomson, 2019-05-08:
Now that would be interesting.. certainly an NMEA output would work very well!

Thinking about it.. what would be amazing is somehow outputting the actual protocol used by the trackers direct.  (emulating the audio modem).  No idea if that is even possible!

digitalentity, 2019-05-08:
I've opened a feature request for NMEA telemetry output. I believe we can close this issue now.

closed 2019-05-10 by shellixyz