iNavFlight/inav #4689: return to home turn direction

MrLAtt, 2019-05-06:
Currently when i actvate RTH the plane turns to the right, however, my antenna is pointed down at 45* and to the left. this means that when at long distance when RTH is activated my groundstation antenna is in the null of my vtx antenna making my video terrible for a few seconds. 

Is there any way to reverse the turn direction for return to home?

Unfortunantly its not really possible for me to change the possition of my antenna since its glued into the end of my starboard wing

shellixyz, 2019-05-06:
It doesn't always turn to the right it turns towards the shortest angle to home

MrLAtt, 2019-05-06:
Oh, ok then, that certainly makes more sense, i guess i must have been slightly off to the right on all the times i tried it today. thanks for the help!

closed 2019-05-06 by shellixyz