iNavFlight/inav #4690: iNAV2.1: Matek f722-SE GPS on UART1 is not working

Ravid824, 2019-05-06:
When connecting GPS to UART1, there is any conflict while the configurator is reacting very slow.
Looks like a conflict with the MSP protocol.
Running iNAV 2.1
Set GPS on UART1 with 57600 baud and enabled the GPS in configuration, UBLOX mode. No other UART ports are enabled beside of the MSP.
Matek recommend to connect the GPS to UART1 (which makes sense since it is close to the SDA/SCL pins which I connected and is working fine) 
It is possible to connect the GPS on UART1 with iNAV2.1? it is a known bug?

Ravid824, 2019-05-06:
BTW, the GPS works fine on UART3.

teckel12, 2019-05-07:
@Ravid824 I'm running Matek F722SE with GPS on UART1 without any issue. Maybe you have a bad UART?

Ravid824, 2019-05-07:
The GPS has compass connected to SDL/DSA and the compass works fine. 
On UART1 the CPU load looks to be very high, GUI almost not moving.
On UART3 it works more or less while after certain time the board is "paused" (leds activities stop) and it releases and continue.
On UART4 is just work perfect.......
It might be related to the GPS configuration? I and using NEO-M8N (for Pixhawk) set the GPS to UBX protocol, 57600 baud.

shellixyz, 2019-05-08:
@Ravid824 Did you disable MSP on UART1 ?  Please post a CLI diff as @teckel12 suggested.

Ravid824, 2019-05-08:
My bad, checked again and when GPS is set on UART1 the CPU load is only 2%.
Yes, MSP is disabled when set GPS on UART1
This is the serial configuration when GPS is set on UART1:
# serial
serial 20 1 115200 115200 0 115200
serial 0 2 115200 57600 0 115200
serial 1 0 115200 115200 0 115200
serial 2 0 115200 115200 0 115200
serial 3 0 115200 115200 0 115200
serial 5 0 115200 115200 0 115200

Attached two DIFF files (one when set GPS on UART1 and one on UART4)
[Matekf722-SE_CLI_DIFF-UART4.txt](https://github.com/iNavFlight/inav/files/3159045/Matekf722-SE_CLI_DIFF-UART4.txt)
[Matekf722-SE_CLI_DIFF-UART1.txt](https://github.com/iNavFlight/inav/files/3159046/Matekf722-SE_CLI_DIFF-UART1.txt)

teckel12, 2019-05-12:
@Ravid824 Try leaving GPS on UART1 but disable PWM output, RSSI  ADC & pitot. If that works, try to enable each isolate.

JohnDevine, 2019-05-13:
Exact same problem here with same hardware setup (running on Mac mini though not windoze). Moving to Uart 3 fixed performance issue.

Ravid824, 2019-05-13:
I sent by board back to shop for exchange (just to make sure that I don't have a faulty board, which I don't believe is the case). Tim, I will  run your recommendations once I will receive the new board.
John, Uart3 it working much better than Uart1, but still timeout from time to time. 
Move the GPS to uart4, it is very stable.

teckel12, 2019-05-13:
My GPS is on UART1 and I have zero problems.

Ravid824, 2019-05-24:
Guys, received the new board. The GPS works well on UART1, it looks that I had a faulty board.
Please close the case. Thank you all for the help and support.

closed 2019-05-31 by digitalentity