iNavFlight/inav #4695: No OSD on matek F722 firmware 2.1.0

Rano85, 2019-05-09:
Hello, my Matek F722 works on fw 2.0.0, OSD OK but I have no OSD on fw 2.1.0 even if I upload font and activate osd feature. In more, I cannot modify resource using CLI. My matek F722 works well with betaflifht OSD and CLI resource

MartinHugh, 2019-05-09:
Just checking, have you tried reflashing, and, doing so with full chip erase ?

DzikuVx, 2019-05-09:
Working fine over here with F722. Remember to set correct mode: PAL/NTSC to match the camera setting

Rano85, 2019-05-09:
Thanks for help. Yes I Flashed with full chip erase. If I unplugged the camera the does not work. I tried Auto, PAL and NTSC > Not ok. The OSD works well only with fw 2.0.0 and betaflight FW. I don't understand. Another issue, on betaflight I can you resource motor 5 none but it is not possible with firmware Inav

MartinHugh, 2019-05-09:
Sorry for the basic suggestion. But check that you have the correct target binary, and, try downloading a new copy and flashing that.

Rano85, 2019-05-09:
Ok thanks good idea in fact, I flashed again with the fw 2.1.0 but without full chip erase and it is work thanks. Do you think it is a good solution or it is a pb to do that

Rano85, 2019-05-09:
And do you have ideas to ressource my 2 servo gimbal and pad S5 and S6?

Rano85, 2019-05-10:
Feature resource je bof availble on inav

closed 2019-05-10 by Rano85