iNavFlight/inav #4697: Quad will not fly straight in position hold.

evmoius, 2019-05-10:
## Current Behavior
When the quadcopter is in position hold mode it will always drift to the left when attempting to fly in a forward straight line. This doesn't happen in horizon or arco mode.

## Steps to Reproduce

1. Arm quadcopter and take off in any mode.
2. Engage position hold (Position hold and altitude hold together) and wait for quad to stop and settle down.
3. Push right stick forward and attempt to fly in a straight line.
4. Quad will noticeably drop front left motor and drift to the left and forward instead of flying straight forward.

## Expected behavior
I expect the quad to fly in a straight line like it does in all other modes.

## Suggested solution(s)
I've been trying to track down a solution or someone who is having a similar problem for this but have come up empty handed.

## Additional context
This is a 500 size quadcopter running 2216 800kv motors and 9x4.5 props.

https://pastebin.com/6U9vejmp

---

- FC Board name and vendor: Matek F405 CTR
- INAV version string: INAV/MATEKF405_SERVOS6 2.1.0 Feb 25 2019 / 16:58:51

shellixyz, 2019-05-10:
PH with `nav_user_control_mode = ATTI` is just like angle/horizon when you move the sticks so it is strange. To me either your quad is pushed by the wind or your board alignment is not correct

evmoius, 2019-05-10:
@shellixyz
I believe I'm using CRUISE at the moment so I can use braking. Should I switch to ATTI?

My goal with this build is trying to make it fly the same way a Phantom does so I can take smooth video.

shellixyz, 2019-05-10:
In your config there is `nav_user_control_mode = ATTI`. If you want it to behave more like a Phantom you need it to be set to `CRUISE` yes.

evmoius, 2019-05-10:
> In your config there is `nav_user_control_mode = ATTI`. If you want it to behave more like a Phantom you need it to be set to `CRUISE` yes.

Wow that's strange. I swear I had it set to cruise already...that explains why braking seemed to be a bit off.  I will set it back to cruise mode and go from there.  Unfortunately I won't be able to fly this build again till next week.

Thanks for spotting that!

DiscoMan18, 2019-05-13:
Could possibly be the same issue with wrong automatic determining of magnetic declination. To make my quad fly straight in poshold CRUISE mode, I had to switch off auto mag decl and to manually specify mag decl of a place where I currently am. Here is the issue I posted:  [https://github.com/iNavFlight/inav/issues/3827](url) There was no feedback from developers.

evmoius, 2019-05-15:
> Could possibly be the same issue with wrong automatic determining of magnetic declination that I faced last summer. To make my quad fly straight in poshold CRUISE mode, I had to switch off auto mag decl and to manually specify mag decl of a place where I currently am. Here is the issue I posted: [https://github.com/iNavFlight/inav/issues/3827](url) There was no feedback from developers.

I'll look into trying this as well once the weather clears up around here...rain in May in California, that doesn't happen often :/

shellixyz, 2019-05-15:
@DiscoMan18 I reopened your issue. Please take a look.

evmoius, 2019-05-21:
> Could possibly be the same issue with wrong automatic determining of magnetic declination that I faced last summer. To make my quad fly straight in poshold CRUISE mode, I had to switch off auto mag decl and to manually specify mag decl of a place where I currently am. Here is the issue I posted: #3827 There was no feedback from developers.

The weather has been absolute crap for flying but I finally managed to get it in the air today. After disabling auto mag decl and entering it manually it seems to be tracking much better than before.
I could easily guide it right where I wanted it to go which I couldn't do before. Unfortunately the weather was still far from ideal with wind gusting to 20mph...I'll have to wait for a better day to really give it a try.

closed 2019-08-03 by stale[bot]