iNavFlight/inav #4710: Crossfire vSpd telemetry

robthomson, 2019-05-15:
The current implementation of csrf telemetry sends the 'ALT' value to the radio via the CSRF protocol.

It would be fantastic to additionally send the vSpeed value.

This information is present in the OSD, so should be possible to send as an additional sensor value.

Its important to have this information for the vario in opentx to function well.

Using ALT is simply too low a resolution for it to be worth using.

teckel12, 2019-05-15:
@robthomson I believe first Crossfire would need to implement it. Contact TBS and make your request.

shellixyz, 2019-05-15:
Yes it is just not supported by Crossfire. You should open a new ticket and ask them here: https://team-blacksheep.freshdesk.com/support/tickets/new

robthomson, 2019-05-16:
Ticket has been raised...

teckel12, 2019-05-16:
@robthomson While on the topic, I'd also like to see TBS add HDOP, distance from home, and fix the YAW sensor that has an overflow bug: https://github.com/iNavFlight/LuaTelemetry/issues/247

robthomson, 2019-05-18:
Looks like TBS are going to add the feature in... but not timeline yet.

I will nudge them every couple of weeks :-)

================
Hi Rob,

Our engineer has included this to the feature request list but we they can't give when it will be implemented. But it is already in the to do list. 

Best regards,

Sherell
Customer Service Representative
TBS Avionics Co Ltd.
www.team-blacksheep.com
==================

teckel12, 2019-05-18:
The 2.95 release is supposed to fix a few telemetry bugs (in beta now). When it's released I'll see if the Yaw bug is fixed.

Probably should close this issue as it's impossible to add to INAV now and there's no time frame on when/if it will be added. Once added to CRSF, a new issue can be opened.

robthomson, 2019-05-18:
Pretty sure this is no longer an issue.. But a feature request? 

Give it a week.   I will have a timeline for you :)

shellixyz, 2019-05-18:
We can leave it open. We will implement it when it is possible to do so

robthomson, 2019-05-20:
Right.  Trappy & Remo have been in touch with me and are happy to add this in.  

Remo has been asking for a suitable DEV on the inav front to liaise with.

Any takers?

In the mean time I will have a chat with Bertrand from OpenTX to see if anything is needed on that front.

wx4cb, 2019-05-20:
Thats awesome. I wish they could fix their rfmd issue lol. Having this would be an awesome addition

sirPerna, 2019-06-03:
Hi

This is Remo from TBS. We are happy to add the support for this. Crossfire firmware doesn't need to be changed to get this working but I will add it to the CRSF specs. 

Would be +/-12.8m/s enough or should we go for a 16bit entry? 
[image]

shellixyz, 2019-06-03:
Hi Remo,

12.8m/s max seems a bit limited to me, please switch to 16bit if possible. I will add this new frame into iNav.

sirPerna, 2019-06-03:
> Hi Remo,
> 
> 12.8m/s max seems a bit limited to me, please switch to 16bit if possible. I will take care of adding this new frame into iNav.

Like this? 
[image]

shellixyz, 2019-06-03:
> Like this?
> [image]

Perfect

shellixyz, 2019-06-03:
@sirPerna Please tell me when a new firmware is available with 16bit vario payload so that I can test #4786

sirPerna, 2019-06-03:
This should already work with any crossfire version >=2.88. Crossfire is only forwarding the frame.

shellixyz, 2019-06-03:
The new sensor will be present in the next release. Tested and merged, closing.

For now you have to build your own OpenTX with this patch for the new sensor to appear on the radio: https://github.com/opentx/opentx/pull/6476

PR #4786 (2019-06-03): Add vario sensor to CRSF telemetry
Closes #4710 

Note: probably won't work correctly with current CRSF firmware versions (<=2.95)
closed 2019-06-03 by shellixyz