iNavFlight/inav #4711: Implement Jeti Ex Bus Code from BF?

RoadyFPV, 2019-05-16:
Is it possible to implement the Jeti Ex Bus Code from BF to Inav?

In BF Ex Bus works great with Telemetry.

shellixyz, 2019-05-16:
What's the issue with the JetiEx in iNav ?

RoadyFPV, 2019-05-16:
I don't recieve telemetry values

bangkokpete, 2019-05-26:
Yes, great feature. all telemetry values are send back to the jetimtransmitter. (incl. GPS data etc.)would be super, if we could get this for INAv as well. Connection via TX2 on my F722SE for telemetry send down to transmitter.
thanks
Peter

RoadyFPV, 2019-07-30:
Hi,
have a coder time to implement this before it closed from bot?
The code works great for telemetry. I use it on betaflight for month without any issue.

It would be nice to have telemetry in inav.

teckel12, 2019-07-30:
@RoadyFPV It may not only be time to do it.  It could also be the lack of hardware to actually test the code.  It typically doesn't work well to try to code something without the hardware to test.  therefore, this will probably require someone with the hardware to write the code (which includes porting it from Betaflight).

RoadyFPV, 2019-07-30:
Testing would not be a problem. Unfortunately, I can not write any kind of code.
Hardware i can to test is Omnibus F7 v2 and Matek F722se.

Maybe that would be helpful

teckel12, 2019-07-30:
@RoadyFPV The hardware really needs to be in possession of the programmer to develop.  It makes any task 100x more complicated when the hardware isn't available to the developer.  This is what I mean by "lack of hardware to actually test".  This is probably one of those issues where someone with this hardware needs to be inspired to write the code.  Shy of that, it's not likely.

closed 2019-10-12 by stale[bot]

Christoph-Hofmann, 2020-11-10:
a friend of mine has the same Problem, the telemetry is not working :-( on jeti