iNavFlight/inav #4719: support for: MATEK F411 one   SPI receiver

X-FOOT, 2019-05-17:
## Current Behavior

## Desired Behavior

## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

X-FOOT, 2019-05-17:
i would like to ask for support for the "matek F411 one" flightcontroller. it has a integrated SPI receiver witch works with the Frsky_X protocol. Betaflight 4.0 has it already integrated in there firmware. I hope someone can do that for inav too, so i can use this board.

X-FOOT, 2019-05-17:
also Frsky_D protocol (taranis D8 mode) for spi receiver. As this works the best

diego0815, 2021-02-01:
Are there any news regarding this feature request? It sounds like a good feature to me, especially also for fixed wings.

But it seems the SPI framework in inav seems to be incomplete right now?

closed 2022-05-07 by DzikuVx