iNavFlight/inav #4720: R9M creates Failsafe in Inav

Basti77, 2019-05-18:
Hello.
since I changed the 868Mhz INAV knows a failsafe although the receiver still has enough reception.
The failsafe LED is not lit.
It doesn't matter which failsafe setting is selected on the receiver. even FS deactivated.
INAV seems to decide for itself.

why does INAV come up with the idea of going into failsafe mode?
I use a Frsky X-lite + R9 Lite module.
Receiver R9 Slim+
I tried EU LBT and Flex with the latest firmware. 190201
there seems to be no difference.

FlightController F35 with INAV 2.1 Relaease.

with a 2G4 receiver on the same model there is no reasonless RTH.

thank you very much for your help
Greetings Sebastian

Basti77, 2019-05-18:
Where does Inav get the RSSI value if I don't use a PPM channel? 
I to see the difference one puts RSSI on a channel.
the transmitter has been set to RI mode I have gone further. I can still move the servos without any problems. the FS LED on the receiver is still off.
[image]
[Desktop.zip](https://github.com/iNavFlight/inav/files/3194061/Desktop.zip)

digitalentity, 2019-05-18:
INAV gets failsafe signal only from the receiver. If receiver is connected via SBUS - there is a special flag in SBUS that indicates the signal loss. INAV trusts the receiver here - it doesn't make random decisions - if receiver is telling INAV that it has a signal loss - INAV goes into failsafe.

Basti77, 2019-05-18:
I found this vid: https://www.youtube.com/watch?v=Gh6i1kyxY6s 
from the release date I installed the last FW available at that time. 

**tata of the failsafe INAFS now corresponds to the one of the recipient.**
but now there are the bugs that were fixed in the following versions. The model is in the basement, the transmitter is in the range check outside the house and I have an RSSI of 48. In the tumble dryer there is also no failsafe at INAV. 
another bug in the Frsky FW?
-

DzikuVx, 2019-05-25:
R9 and R9M are working fine with INAV. I have several quads and airplanes on that configuration.

closed 2019-05-25 by DzikuVx