iNavFlight/inav #4721: OmnibusF4 V3 baro not detected 2.2.0 development branch.

rivermtnn, 2019-05-19:
I compiled the firmware for OmnibusF4V3 2.2.0 development branch source. The baro is not deteced anymore on this version. When flashed with 2.1.0 firmware loaded from the configurator, the baro is detected as BMP280 without problem.

The board is a generic OmnibusF4V3 which is widely available.

Some CLI output as below from this 2.2.0 firmware:

# version
# INAV/OMNIBUSF4V3_SFTSRL2 2.2.0 May 19 2019 / 21:47:32 (dfd0b62)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

# status
System Uptime: 451 seconds
Current Time: 2019-05-19T14:26:22.292+00:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000, MAG=IST8310
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=NONE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
Stack size: 6144, Stack address: 0x10010000, Heap available: 1928
I2C Errors: 0, config size: 5184, max available config: 131072
ADC channel usage:
BATTERY : configured = ADC 2, used = ADC 2
RSSI : configured = ADC 3, used = none
CURRENT : configured = ADC 1, used = none
AIRSPEED : configured = none, used = none
System load: 2, cycle time: 1003, PID rate: 997, RX rate: 49, System rate: 9
Arming disabled flags: COMPASS ACC RX CLI

# resource
IO:
----------------------
A00: FREE 
A01: FREE 
A02: FREE 
A03: FREE 
A04: MPU CS
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: FREE 
A09: SERIAL1 UART TX
A10: SERIAL1 UART RX
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: OSD CS
B00: FREE 
B01: FREE 
B02: FREE 
B03: FLASH CS
B04: BEEPER OUT
B05: LED1 OUT
B06: FREE 
B07: FREE 
B08: FREE 
B09: FREE 
B10: I2C2 SCL
B11: I2C2 SDA
B12: FREE 
B13: FREE 
B14: PPM IN
B15: FREE 
C00: INVERTER OUT
C01: FREE 
C02: ADC CH2
C03: FREE 
C04: MPU EXTI
C05: FREE 
C06: FREE 
C07: FREE 
C08: FREE 
C09: FREE 
C10: SPI3 SCK
C11: SPI3 MISO
C12: SPI3 MOSI
C13: FREE 
C14: FREE 
C15: FREE 
D00: FREE 
D01: FREE 
D02: FREE 
D03: FREE 
D04: FREE 
D05: FREE 
D06: FREE 
D07: FREE 
D08: FREE 
D09: FREE 
D10: FREE 
D11: FREE 
D12: FREE 
D13: FREE 
D14: FREE 
D15: FREE 

Hope this is a small issue and could be easily fixed.

rivermtnn, 2019-05-19:
After a bit study of the source, found out that because of OMNIBUSF4V3_SFTSRL2 is new target variant, the target.h file was not updated thoroughly to take it in, the baro was left out if this target was chosen when compiling.

So I edit the target.h a bit and now it's working again.

Please update that file, thanks.

shellixyz, 2019-05-26:
I just saw your issue. Unfortunately we already released 2.2-RC1. It will be fixed in 2.2-RC2

PR #4747 (2019-05-27): Fix OMNIBUSF4/target.h
Closes #4721
closed 2019-05-28 by DzikuVx