iNavFlight/inav #4735: Waypoint and RTH failure..or my mistake!!!

Gabbocombinaguai, 2019-05-24:
Hello
Im going to describe as best as i can do and share a link about an issue accurred using  WP mode with matek f411
Look at video min. around min 2.20          https://youtu.be/BQ3DHl4vxBQ

I loaded the WP mission on the Fc. , first wp was 20 m away from where I was and thesecond was 1 km towards east and 150m. high
During the ascend i was wearing my glasses so i was non paying attention at the screen. When i was 500m away the signal between the plane and the radio was lost so the plane went into RTH mode; but after a second the link was restored and here was the problem. RTH was off, wp was off and the plane was pointing the ground and i was away from screen. Hopefully after few seconds i sow the situation and i grabbed the radio to take control of the plane.

So now im asking help regarding this behaviour. Was that normal?
Thanks.

wilco1967, 2019-05-26:
I guess when you had a failsafe, the WP mission got interrupted and it went into RTH. 
When the link returned, you probably must have wiggled the sticks, canceling RTH, and (I assume) your WP switch was still active, so the FC started the WP mission from the beginning, So it immediately try to get towards the first WP, at whatever altitude you set WP1 at, at maximum down angle.. (assuming WP1 altitude was set lower then you were currently flying).

that is normal behaviour.

Gabbocombinaguai, 2019-05-26:
The remote control was on the table, it was not in my hands. Moreover the first WP was not in that position but just 15 m away from my car and 15 m high. As you can see from the video the plane was pointing towards the ground at least 100 m far from first WP.

Thanks anyways

closed 2019-08-08 by stale[bot]