iNavFlight/inav #4736: (WING) Transition from ALTHOLD to RTH back to ALTHOLD causes steep dive

b14ckyy, 2019-05-24:
## Current Behavior
If RTH Altitude is above current ALTHOLD altitude (onlyusing this in CRUISE mode) the plane will climb to RTH altitude as expected. Switching back to CRUISE+ALTHOLD causes a steep dive back to prev. ALTHOLD level

## Steps to Reproduce

1. Set RTH altitude to 200m
2. Fly in CRUISE mode with ALTHOLD added
3. Switch to RTH or trigger Failsafe
4. wait till climb to 200m is done
5. disable RTH or turn on radio again
--> Plane will go into a steep dive (not sure about angle but looks even steeper than nav_fw_dive_angle to go back to previous height of ALTHOLD

## Expected behavior
When FS or RTH is disabled and switched back to ALTHOLD, the plane should keep it's actual level. 

## Suggested solution(s)
Reset ALTHOLD height when switching from RTH back to ALTHOLD

## Additional context
This behavior can be dangerous when doing low flight in ALTHOLD mode like 10-20m over ground. Triggering FS and then taking over control again, will cause a fast dive that can overshoot and crash the plane in the ground, especially if the ground elevation is higher than on the previous position. 

If a log is needed, I need to go for a flight again the next days. 

---

# INAV/MATEKF405SE 2.1.0 Feb 25 2019 / 17:02:18 (65b0ec142)
# GCC-7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

digitalentity, 2019-05-24:
Yes, log is necessary to understand what the firmware is doing.

b14ckyy, 2019-05-24:
I'll ask someone in the FB group to test this as I'm not sure if I will go out with my AR wing this weekend.

wx4cb, 2019-05-24:
do you have DVR too if possible as this doesn't make sense. if you're in failsafe because the tx is off i can't see it why it wouldn't reset to whatever mode it was in before it failsafed.

b14ckyy, 2019-05-24:
the problem is, if ALTHOLD was active at a lower height than RTH mode, it will climb to the height that is set up for RTH. But when switching back to ALTHOLD directly, it will immediatly descent back to the previous ALTHOLD height instead staying at the height it had during RTH. Maybe this will also happen if it was very high and RTH is set to fixed height but never tried that. Let me check my FPV recordings. Maybe i will find a log too. I have about 200 log files on my HDD

wx4cb, 2019-05-27:
i've just tested this on two different airframes and it all works as expected using 2.1 

DVR Is below, First one is the S1100 twin, Second one is the Mini Drak. Both have an RTH height set to 200 feet. I should have set it higher, but i didnt have my laptop with me to change it.

NOTE, These are raw dvr so make sure you're not wearing headphones :D

https://youtu.be/rGyD0ciKA-Y

https://youtu.be/lTxr0v0vL9w

ShawnzAZ, 2019-05-28:
I tested this yesterday as well.   Returning to cruise when in RTH just held the heading and altitude it was in while executing the RTH command.  It did not try to return to previous the Cruise / Altitude Hold altitude.

If it matters it was a conventional airplane (Bix 3) with flaps all run through the inav mixer vs a flying wing.

Inav 2.1 on F722-Wing (MATEKF722SE target)

b14ckyy, 2019-05-28:
I will go fly on Thursday, to try a new prop. I will try to reproduce then with bbx and fpv recording

wx4cb, 2019-05-28:
I seem to remember that there was a conversation a while back about making sure your rth and home altitude wasnt the same. I would also check that

ShawnzAZ, 2019-05-29:
@b14ckyy , this is the DVR from my testing, to see if what I did should have re-created the issue:

https://youtu.be/yU2Y11sAux8

wx4cb, 2019-05-29:
might want to increase your climb throttle, i was in anticipation to see whether it would stall or not lol

> @b14ckyy , this is the DVR from my testing, to see if what I did should have re-created the issue:
> https://youtu.be/yU2Y11sAux8

b14ckyy, 2019-05-29:
wanna look into my tuning guide? :P 
jk
Flight day tomorrow is fixed. Will report in the evening. Need some tuning anyway as I switched to 8045 blade to see if i can squeeze out some more flight time :)

ShawnzAZ, 2019-05-30:
Haha it was pretty windy (20+ MPH gusts) and that Bix3 is pretty much a motor glider.  It's all good :)

b14ckyy, 2019-06-02:
I was not able to reproduce yet. As I want to try out 2.2 asap and already updated, I'm not sure if I will see this again. Maybe we close This issue for now. If a see that again i will save the log and DVR and open a new one.

closed 2019-08-15 by stale[bot]