iNavFlight/inav #4744: Mistake in documentation?

EnsignCrasher, 2019-05-26:
Was reading through this page: https://github.com/iNavFlight/inav/wiki/Tune-INAV-PIFF-controller-for-fixedwing

In the end, there is a short abstract telling: 

> If your plane over corrects when RTH is engaged, try increasing nav_fw_pos_xy_p and/or increasing nav_fw_pos_xy_i. Good values to start: "set nav_fw_pos_xy_p = 50"; "set nav_fw_pos_xy_i = 5". Also you can lower "nav_fw_pos_xy_d". The behaviour of the plane is very different with or w/o wind, so it is necessary to test and tweak parameters in both scenarios.

Shouldn't that be, "if your plane corrects not enough when RTH is engaged, try increase...."?
Was just a little confusing to me

digitalentity, 2019-05-26:
No, this is correct. If the plane overshoots when turning (symptom is a wave-like flight path) - it means there is not enough P and I to turn. When P & I are too high the symptom is fast wandering left-right by a small amount (less than 5 deg).

EnsignCrasher, 2019-05-26:
Ah, very good explanation :)
I would say, copy-pasteable for the documentation!

EnsignCrasher, 2019-05-26:
If its ok for you, i can make a PR from it

digitalentity, 2019-05-26:
Absolutely!

EnsignCrasher, 2019-05-27:
ah didn't even need a PR for wiki pages ;)

closed 2019-05-27 by EnsignCrasher